MarlinFirmware/Marlin #16251: All nine G29 probes happen in the same place

BearcatTd, 2019-12-18:
When I run a G29 command all nine probes occur in the same place...lower left corner of printbed

### My Configurations
[Immel Configuration H and adv H.zip](https://github.com/MarlinFirmware/Marlin/files/3979605/Immel.Configuration.H.and.adv.H.zip)

Printer: Ender3
Board: SKR E3 mini V1.2
Interface:  Raspberry PI 3 B+ / OctoPrint
Hotend/Extruder: Aero Titan
Probe: BLTouch

1. Power up printer
2.  Open Octoprint in web browser
3.  Run G28 
4.  Run G29
5.  Error occurs
6.  Power off printer

**Expected behavior:** Probe nine points around the printbed

**Actual behavior:** See above and video below

#### Additional Information

[IMG_3569.mov.zip](https://github.com/MarlinFirmware/Marlin/files/3979608/IMG_3569.mov.zip)

tatusah, 2019-12-18:
Try disabling `MONITOR_DRIVER_STATUS` and also probe offset settings are missing from your config.

BearcatTd, 2019-12-18:
Thanks for the response Tatusah!  

MONITOR_DRIVER_STATUS is already //disabled. 

in Configuration.h these are the only "PROBE_OFFSET" I could find.
#define NOZZLE_TO_PROBE_OFFSET { 21, -44, -1.46 }

// For M851 give a range for adjusting the Z probe offset
#define Z_PROBE_OFFSET_RANGE_MIN -20
#define Z_PROBE_OFFSET_RANGE_MAX 20

Thanks again, this has been going on for about two weeks now...

tatusah, 2019-12-18:
Ok, it was just that the configuration files in the zip you uploaded here had the MONITOR_DRIVER_STATUS enabled. Also the NOZZLE_TO_PROBE_OFFSET values were zero in the uploaded configs. So when you upload the configs here make sure that they are actually the ones you are using and causing the issues.

Does the homing and bed leveling work through LCD or is this only happening with Octoprint? Also make sure that you are using the most recent bugfix-2.0.x branch firmware. There was just a while ago a bug that caused issues with bed leveling if using decimal numbers with nozzle to probe offset. So you could also check if the problem goes away with updating the firmware and trying to change the offset values to integer numbers.

Anyway, this seem to be a configuration issue and not a bug. I also have the same SKR board as you and it's doing bed leveling without issues. If the above doesn't help then perhaps you should get help for configuring the printer is through: 

-RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
-[Tom's 3D Forums](https://forum.toms3d.org/)
-Facebook Group "[Marlin Firmware](https://www.facebook.com/groups/1049718498464482/)"
-Facebook Group "[Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/)"
-[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
-Marlin Discord server. Join link: [https://discord.gg/n5NJ59y](https://discord.gg/n5NJ59y)

That way you can get help "realtime" and don't have to wait for back and forth messaging through github.

tatusah, 2019-12-18:
Also basic things that should always be done after flashing firmware is to initialize EEPROM either trough LCD or with M502 & M500 commands that you can input in octoprint terminal. Forgetting this step is also known to cause many kinds of issues.

BearcatTd, 2019-12-19:
I'm reflashing everything and will give it another go.  I'll upload the new files if needed.  I apologize for uploading the wrong ones...

closed 2019-12-19 by BearcatTd