MarlinFirmware/Marlin #16255: [FR] Independent baudrate setting for secondary serials

karabas2011, 2019-12-19:
I want to connect different serial devices to my boards 
some of them are 9600, some 115200.
For Pronterface-USB connection I use 250000
But marlin 2.0 does not allow to set  separated baudrates for secondary serial.
Any hack?

ellensp, 2019-12-19:
You can just edit Marlin/src/Marlin.cpp find MYSERIAL1.begin(BAUDRATE); and change BAUDRATE to be a number..    (Some ofther board have it set elsewhere eg Marlin/src/HAL/HAL_LPC1768/main.cpp

ellensp, 2019-12-19:
Or add the line #define BAUDRATE1 9600 to your configuration.h and change MYSERIAL1.begin(BAUDRATE); to MYSERIAL1.begin(BAUDRATE1);

karabas2011, 2019-12-19:
Big thanx! Last question 
I use ESP wifi on secondary port and want to connect third device on 9600 on spare pins.
Is it possible? ( I need only TX pin no RX from device)

ellensp, 2019-12-19:
Sure it can be done... but not without coding... The firmware would need to know what is on the serial port in order to talk to it.  Also if its a 8 bit controller, it pays to use real hardware serial ports.

karabas2011, 2019-12-23:
> Sure it can be done... but not without coding... The firmware would need to know what is on the serial port in order to talk to it. Also if its a 8 bit controller, it pays to use real hardware serial ports.
For low baudrate devices I used software serial lib ( for arduino) a couple years ago.  It worked on any spare pins. Now for Marlin 2.0 I did not try this lib.

sjasonsmith, 2020-09-11:
I think this is closely related to #17599. The more ports avaialable, the more important this becomes.

slowbro, 2021-07-03:
#21949 was merged, this can be closed.

closed 2021-07-03 by thisiskeithb