MarlinFirmware/Marlin #16259: [BUG] Bed Heating "undershoots" by ~8°C

Seref, 2019-12-19:
### Bug Description
When setting the heated bed to 60°C it normaly rises to 52°C and then stops or rather holds the temprature, this triggers an error because the temprature isn't increasing. The overall system does reach 60°C when PID-Autotuning and also hold it there (I'm taking eight measurements).

### My Configurations
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3983969/Configuration_adv.zip)
I'm using an SKR Mini V1.2 E3 with an MK2B heated bed and a 100K NTC Thermistor with Marlin bugfix-2.0.x 

### Steps to Reproduce
1. Set the temp in Pronterface to 60°C
2. Wait

**Expected behavior:** It delivers "full power" until 60°C and then starts to hold the temperature

**Actual behavior:** Around 50-52°C it starts to hold the temperature

#### Additional Information
[image]

ellensp, 2019-12-19:
This is often hardware.   Possible cause are: Thermistor not secure on the bed, Bed wires not thick enough to power bed. Bed wires not securely fastened at controller. Power supply to week not providing enough current. No insulation under the heated bed.

Seref, 2019-12-19:
Well I thought so too and replaced the Thermistor with a new one, added to that I also have insulation under my heated bed and like mentioned before when PID-autotuning it does reach 60°C without a problem and also holds it for eight measurements.

Lord-Quake, 2019-12-19:
What happens with Marlin 2.0.0 ?

Seref, 2019-12-19:
Okay well with 2.0.0 (same config files) it works without a problem!
[image]

Seref, 2019-12-19:
Okay the latest [bugfix-2.0.x](/MarlinFirmware/Marlin/tree/bugfix-2.0.x) fixes the problem.

closed 2019-12-19 by Seref