MarlinFirmware/Marlin #16261: [BUG] Marlin 2.0 + BLtouch +Anet A8 = not enough space

mateusz3233, 2019-12-19:
It looks like Marlin 2.0 is to big to implement with BLtouch to Anet A8

Does anyone knows what can be removed from the code to put into my printer ?
I followed steps to activate BLtouch via this intruction:
`https://www.youtube.com/watch?v=WWDkZtWwd6I`

bill-orange, 2019-12-19:
I removed SD support since I use OCTOPI.  Are you using bed PID?  It's unnecessary and takes up a surprising amount of space.  You could load OPTIBOOT with a USBASP but that's a pain at least for me since the printer is in another part of the house from the computer.

Lord-Quake, 2019-12-19:
This is not a Marlin bug.
You will have to look for options within the code that you will not be needing in order to reduce the firmware size. For example by deactivating ARC_SUPPORT.

Vertabreak, 2019-12-19:
using platformio will also greatly reduce the size i did a video showing the same code compiled with vscode and Arduino in vscode it was 1kb less ram and 6kb less progmem.

bill-orange, 2019-12-19:
@Vertabreak can you provide a link to that video?

Vertabreak, 2019-12-19:
can do.
https://youtu.be/oUG5kweG07U
note that my channel is geared to my fork that is different then base marlin but is based on it.

boelle, 2019-12-19:
closing, not a bug

closed 2019-12-19 by boelle

Vertabreak, 2019-12-19:
just need the old shoe horn to make it fit lol.