MarlinFirmware/Marlin #16262: [FR] Add adjustable debounce/key repeat delay for ADC buttons

soligen2010, 2019-12-19:
### Description

Add adjustable debounce/key repeat delay for ADC buttons

### Feature Workflow

This is a feature I have implemented locally.  As I am a first time contributor, I created this issue to be sure you are OK with a pull request for the change.

Buttons that use a single ADC pin have a ADC count for de-bounce and key repeat hard coded to 16.  I propose making this an item in the configuration do this can be adjusted.

#### Additional Information

I found increasing the value from 16 to 32 made my ANET A8 much easier to use.

Affects displays ULTRA_LCD and ZONESTAR_LCD

borland1, 2019-12-19:
Could this FR include debounce adjust for rotary encoder also?

soligen2010, 2019-12-19:
That would be in a totally different place in the code, so that FR would be unrelated to this one.  I don't have a machine with a Rotary Encoder, so I would not be comfortable trying to change that code as I would not be able to test it. (And I personally don't know where that piece of code is).

borland1, 2019-12-20:
I found a code section titled "Additional Features" in Configuration_adv.h.  This advanced configuration section includes a Rotary Encoder step rate (steps/sec) for debouncing.  Note: This section is not to be confused with similarly titled section "Additional Features" in Configuration.h, which includes rotary encoder Menu settings, namely: Encoder-Pulses/step, & MenuItems/step.

closed 2020-06-30 by github-actions[bot]