MarlinFirmware/Marlin #16263: [FR] Adding Anet E10 Configuration files

Bazzofski, 2019-12-19:
### Description

Suggestion : adding configuration files for Anet E10 printers

Vertabreak, 2019-12-19:
Does it need ported over from old version?

Bazzofski, 2019-12-19:
No, but i've made my own config files for Marlin 1.1.x .

Vertabreak, 2019-12-19:
post old config files and ill port them over and submit them.

Bazzofski, 2019-12-19:
Where should I post it ? Sorry, i'm pretty new on github.

thisiskeithb, 2019-12-19:
> Where should I post it ? Sorry, i'm pretty new on github.

Zip up both your `Configuration.h` & `Configuration_adv.h` files and attach them in a reply.

Vertabreak, 2019-12-19:
Just zip them and drop them here in your next message.

Bazzofski, 2019-12-19:
I didn't thought about zipping them, thanks ^^
[Marlin Anet E10 config.zip](https://github.com/MarlinFirmware/Marlin/files/3985004/Marlin.Anet.E10.config.zip)

Vertabreak, 2019-12-19:
I'll run a diff on them shortly check back in a hour or two. Once I have a copy on my fork you can test them.

Vertabreak, 2019-12-19:
k gonna need the full source copy as the config doesn't tell me everything.

Vertabreak, 2019-12-19:
running the last few checks on the updated config should have something ready to test shortly.

Vertabreak, 2019-12-19:
ok so here is a link to the prepped pull request.
https://github.com/Vertabreak/Marlin/tree/PR6

here is a copy compiled on vscode for testing.
[E10.hex.zip](https://github.com/MarlinFirmware/Marlin/files/3985405/E10.hex.zip)

Bazzofski, 2019-12-19:
I can't flash it on my printer right now, I'll try tomorrow and tell you if everything is ok for the pr.

Vertabreak, 2019-12-19:
Not anticipating any major issues just looking for confirmation as all the hardware was existing supported stuff I just set it up based on your original configurations.

Bazzofski, 2019-12-19:
I don't see any alterations to add, you can create the pr. Thanks for your work !

Vertabreak, 2019-12-19:
would like hardware confirmation that it runs on the printer some photos would be good ^^.

Bazzofski, 2019-12-20:
I've tried to flash the firmware but it keeps showing error on compiling about undefined values, I've checked and they're defined in the advenced config file.

Logs : 

`sketch\src\feature\pause.cpp: In function 'bool unload_filament(const float&, bool, PauseMode)':

sketch\src\feature\pause.cpp:348:21: error: 'FILAMENT_UNLOAD_RETRACT_LENGTH' was not declared in this scope

   do_pause_e_move(-(FILAMENT_UNLOAD_RETRACT_LENGTH) * mix_multiplier, (PAUSE_PARK_RETRACT_FEEDRATE) * mix_multiplier);

                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

sketch\src\feature\pause.cpp:348:21: note: suggested alternative: 'FILAMENT_UNLOAD_PURGE_LENGTH'

   do_pause_e_move(-(FILAMENT_UNLOAD_RETRACT_LENGTH) * mix_multiplier, (PAUSE_PARK_RETRACT_FEEDRATE) * mix_multiplier);

                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

                     FILAMENT_UNLOAD_PURGE_LENGTH

sketch\src\feature\pause.cpp:351:14: error: 'FILAMENT_UNLOAD_DELAY' was not declared in this scope

   safe_delay(FILAMENT_UNLOAD_DELAY);

              ^~~~~~~~~~~~~~~~~~~~~

sketch\src\feature\pause.cpp:351:14: note: suggested alternative: 'FILAMENT_UNLOAD_PURGE_DELAY'

   safe_delay(FILAMENT_UNLOAD_DELAY);

              ^~~~~~~~~~~~~~~~~~~~~

              FILAMENT_UNLOAD_PURGE_DELAY

Vertabreak, 2019-12-20:
unsure what you attempted to compile but i posted a fully compiled copy.

Vertabreak, 2019-12-21:
gotten a chance to test yet?

Bazzofski, 2019-12-24:
Sorry for the delay. I tried to compile the hex file but didn't find how to do it, so i tried compiling your pr and I got those errors : 

https://pastebin.com/NrRHYxNK

Then I've tried the official 2.0.x release with the new config files and I get errors too : 

https://pastebin.com/BGvSFy5U

It doesn't seems to come from the config files.

Vertabreak, 2019-12-24:
compiled normal for me unsure what might be the issue on your end.

Bazzofski, 2019-12-24:
I compiled it with vscode using platformio

Vertabreak, 2019-12-24:
but did you set the correct environment for that board.

Vertabreak, 2019-12-24:
[image]

Vertabreak, 2019-12-24:
the default is mega2560 that is not the board you have.

Bazzofski, 2019-12-24:
Yes, I imported the project with ATmega1284P selected and I've changed it too in the .ini file.

Vertabreak, 2019-12-24:
[image]

Vertabreak, 2019-12-24:
[image]

Bazzofski, 2019-12-24:
I didn't change the environment, it set it automatically when I chose the board while importing the arduino project.

Bazzofski, 2019-12-24:
[image]

Vertabreak, 2019-12-24:
the preped copy i posted literally is all setup you just open the folder and click build yea that isn't right.

Vertabreak, 2019-12-24:
[image]

Bazzofski, 2019-12-24:
Sorry, my bad ^^ I was using an other pr. I tried and this time it successfully compiled. But now avrdude won't flash the board with the "not in sync" error.
[image]

Vertabreak, 2019-12-24:
is the right com port selected and are the drivers installed.
[image]

Bazzofski, 2019-12-24:
Yes, it is detected and It has a bootloader because i've already flashed old marlin versions on it.

Vertabreak, 2019-12-24:
its the correct com port? its not trying to flash like another com device on the system.

thisiskeithb, 2019-12-24:
> the default is mega2560 that is not the board you have if i recall its a sanguino_atmega1284p

So we're not troubleshooting a stock Anet E10 config?

Vertabreak, 2019-12-24:
marlin default is mega2560 not talking about the printer.

Vertabreak, 2019-12-24:
config is a direct port of the original.

Bazzofski, 2019-12-24:
It's clearly the good com port, I can access the printer with pronterface with the same port.

Vertabreak, 2019-12-24:
not really troubleshooting more like explaining how to use the tools required to flash the printer.

Vertabreak, 2019-12-24:
maybe the baud rate is set incorrectly?

Vertabreak, 2019-12-24:
unsure why just flashing the pre compiled hex file was the 2ed choice.

Bazzofski, 2019-12-24:
I've changed the baudrate, I'll tell you when it's finished.

Vertabreak, 2019-12-24:
kk finally flashing can get to the real testing.

Bazzofski, 2019-12-24:
It gave me the same error, I'm gonna try flashing it directly with avrdude without vscode.

Vertabreak, 2019-12-24:
i normally use octoprint to flash pre compiled hex files from vscode personally.

Bazzofski, 2019-12-24:
Yes, but I don't have any octoprint set-up

Vertabreak, 2019-12-24:
cura can flash hex files also.

Vertabreak, 2019-12-24:
you didn't happen to have cura open when trying to flash did you or anything else connected to the printer as that could block the connection or even interrupt it mid flash.

Bazzofski, 2019-12-24:
The port was free when I tried. I've tried to flash it with cura but it reset the printer when starting and disconnect it.

Vertabreak, 2019-12-24:
cura auto connects and reboots marlin its just how cura works its kinda dangerous cus if you open cura mid flash you just bricked it.

Bazzofski, 2019-12-24:
That's a bit crappy from cura, did they release your pr ?

Vertabreak, 2019-12-24:
if you really insist on using arduino disable any non vital features and manage to get arduino to compile it. it was only a idea about submitting a pull request to stop the auto connect in cura.

Vertabreak, 2019-12-24:
id hate to make the suggestion but your on windows right have you tried rebooting lol.

Bazzofski, 2019-12-24:
Yep, I'm on windows, I'v already had the reflex to reboot it xD but it didn't change anything. I think it's more of a board config problem, I'm gonna try something with arduino ide

Vertabreak, 2019-12-24:
its all kinda weird cus Ive flashed the entire wrong boards hex file to the wrong printer several times it just doesn't boot up and i have to re-flash the right one.

Bazzofski, 2019-12-24:
Yes, it's really weird, only the prepd vscode project you've sent is compiling. An only the arduino ide can flash the board but it can't compile the project.

Vertabreak, 2019-12-24:
try the melzi environment in platformio

Bazzofski, 2019-12-24:
Trying that and updating you when it's finished ^^

Vertabreak, 2019-12-24:
i was digging around in the code and it says both them are also ATmega1284P.

[16 more replies]

closed 2020-01-24 by boelle