MarlinFirmware/Marlin #16267: [FR] Does marlin support MKS robin2 + TFT display?

cjsoong, 2019-12-20:
https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin2

ellensp, 2019-12-20:
We're not a support group

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

-RepRap.org Marlin Forum
-Tom's 3D Forums
-Facebook Group "Marlin Firmware"
-Facebook Group "Marlin Firmware for 3D Printers"
-Marlin Configuration on YouTube
-Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

shitcreek, 2019-12-20:
@ellensp not to undermine you, but we do allow questions.

There shouldn't be any reason why Marlin wouldn't support the MKS Robin2, but it looks like it hasn't been ported over. I'll pull the stuff off of Makerbase and put together a PR.

shitcreek, 2019-12-20:
@cjsoong please give the PR a test if you've got this board.

boelle, 2019-12-20:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-12-20 by boelle

thisiskeithb, 2019-12-20:
This probably should be reopened and renamed to be a feature request.

boelle, 2019-12-20:
OP could put in a tiny more effort that just a link

or at least give response

boelle, 2019-12-20:
anyways @shitcreek  did make a PR to support the board