MarlinFirmware/Marlin #16273: [BUG] Cannot

randellhodges, 2019-12-20:
### Bug Description

Cannot compile with 2 z steppers and G34 enabled.

```
Marlin\src\gcode\calibrate\G34_M422.cpp:88:24: error: non-constant condition for static assertion
 static_assert(LTEST(2) && RTEST(2), "The 3rd Z_STEPPER_ALIGN_XY X is unreachable with the default probe X offset.");
                        ^
Marlin\src\gcode\calibrate\G34_M422.cpp:88:24: error: array subscript value '2' is outside the bounds of array type 'const xy_pos_t [2]' {aka 'const XYval<float> [2]'}
Marlin\src\gcode\calibrate\G34_M422.cpp:89:24: error: non-constant condition for static assertion
 static_assert(FTEST(2) && BTEST(2), "The 3rd Z_STEPPER_ALIGN_XY Y is unreachable with the default probe Y offset.");
                        ^
Marlin\src\gcode\calibrate\G34_M422.cpp:89:24: error: array subscript value '2' is outside the bounds of array type 'const xy_pos_t [2]' {aka 'const XYval<float> [2]'}
Compiling .pio\build\LPC1768\src\src\gcode\calibrate\M666.cpp.o
*** [.pio\build\LPC1768\src\src\gcode\calibrate\G34_M422.cpp.o] Error 1
```

### My Configurations

https://github.com/randellhodges/Marlin/tree/skr-1.3/Marlin

### Steps to Reproduce

have dual z steppers, enabled g34 and try to compile

**Expected behavior:** Compiles

**Actual behavior:** static asserts prevent compile

#### Additional Information

I think these need to be wrapped in an #if

G34_M22.cpp

```
static_assert(LTEST(2) && RTEST(2), "The 3rd Z_STEPPER_ALIGN_XY X is unreachable with the default probe X offset.");
static_assert(FTEST(2) && BTEST(2), "The 3rd Z_STEPPER_ALIGN_XY Y is unreachable with the default probe Y offset.");
```

Maybe use `#if ENABLED(Z_TRIPLE_STEPPER_DRIVERS)`

sjasonsmith, 2019-12-20:
Do you intend to post a PR for this yourself?

sjasonsmith, 2019-12-20:
This must mean that we don’t have a Travis test for Dual Z?

randellhodges, 2019-12-20:
> Do you intend to post a PR for this yourself?

Yup, just created the bug to reference in the PR :)

boelle, 2019-12-24:
did pr #16274 fix this one?

randellhodges, 2019-12-25:
Indeed that PR fixed this. Closing.

closed 2019-12-25 by randellhodges