MarlinFirmware/Marlin #16275: [BUG] Line 810 in unwarm_thumb.cpp produces a compilation warning.

victorpv, 2019-12-20:
### Bug Description

This line produces an error message on compilation time:
Assignment to itself 'state->regData[rhd].o  = state->regData[rhd].o'	unwarm_thumb.cpp	/src/HAL/shared/backtrace	line 810

[          state->regData[rhd].o  = state->regData[rhd].o;](https://github.com/MarlinFirmware/Marlin/blob/5103cdcff6a2d40c9d7eb5de2274fe23dbf56797/Marlin/src/HAL/shared/backtrace/unwarm_thumb.cpp#L810)

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Try to compile for an STM32F1R board.
2. Error shows during compilation.

**Expected behavior:** [What you expect to happen]
Not an error.

**Actual behavior:** [What actually happens]
Assignment to itself 'state->regData[rhd].o  = state->regData[rhd].o'	unwarm_thumb.cpp	/src/HAL/shared/backtrace	line 810

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

victorpv, 2019-12-20:
Adding @ejtagle  since I believe he contributed that file.

boelle, 2019-12-24:
maybe submit a PR to fix it?

victorpv, 2019-12-25:
I wish I knew the fix, I would have sent a PR, but I really do not understand the function of that code.
The compiler throws a warning, and makes sense that it gives a warning since a variable is assigned back to itself with no change. I went back on that file until I found when that line was added. I suspect the correct line could be this, but really do not know and have no way of checking since I do not know what that section is used for:
`state->regData[rhd].o = state->regData[rhs].o`

boelle, 2020-01-08:
@victorpv does the compile fail or is it just a warning?

boelle, 2020-01-08:
@ejtagle any idea here?

victorpv, 2020-01-08:
@boelle it's a warning, the code compiles. Since the variable is assigned back to itself, which wouldn't make much sense, the compiler throws a warning, and I would think that line has a mistake unless is a very clever programming trick I can't understand. Hopefully @ejtagle can shed some light.

boelle, 2020-01-14:
since its just a warning and we have not heard from @ejtagle i will close this one

if it turns out to be a mistake in code that later give errors we can always reopen

closed 2020-01-14 by boelle