MarlinFirmware/Marlin #16276: [FR] Advanced control from printer depending on layer type

syproduction, 2019-12-20:
### Description

Menu items for controlling movement options depending on type of layer.

### Feature Workflow

Control menu with submenus for each type of layer. There are strings in gcode like ;TYPE:SKIRT so it would be awesome to set different flow or acceleration or fan speed etc. for these types. 

It is very useful if we could manage these parameters on the fly and control visually how good this tuning is.

Vertabreak, 2019-12-20:
adding additional menus costs system resources not to say its not a interesting idea.

ellensp, 2019-12-21:
That's a comment generated by the slicer, comments are not sent to the controller.

AnHardt, 2019-12-21:
Any unconditional statement here is a bit dangerous.
If comments of this kind are produced depends on the 'slicer' and its settings.
The exact format of this comments may depend on the slicers settings.
If the comments are sent to Marlin depends on the host-program.
A comment is a comment and is deleted in a very early state of Marlins command processing.

A clean way to handle this would be.
Post-process the g-code produced by the slicer (maybe in the slicer) to produce g-code from the comments.
Then the host **will ** send something to Marlin and Marlin will not delete but process the code.

At first define the new g-codes (ideally together with the other firmwares, hosts and slicers authors), then write the code implementation for Marlin and the preprocessors.

God luck.

MarkMan0, 2019-12-26:
I don't think gcodes should be used for this. Gcodes are meant to control the printer. This feature is more like - tell the printer what it's actually doing.
Some other style would be better suitable for this, like 

> @TYPE:skirt
..................
@TYPE:wall
@TYPE:retraction

etc.

This way it would be clear, what line is doing what, and the host/printer could simply ignore, if they aren't used, without parsing the line and looking up if the gcode exists.

closed 2020-06-30 by github-actions[bot]