MarlinFirmware/Marlin #16278: [BUG] Anet A6 "garbage" on LCD screen in 2.0 (works in 1.1.9)

Uinden, 2019-12-20:
### My Configurations

Unmodified configs https://github.com/MarlinFirmware/Marlin/tree/2.0.x/config/examples/Anet/A6

### Bug Description

I have Anet A6 printer and I am updating from marlin 1.1.9 to 2.0 and I have LCD problem.
I upload 1.1.9 and it works fine. (I downloaded that version in August 2018)
I upload 2.0 and its unusable. It has "garbage" all over the screen and it randomly changes on each screen refresh.
Tested bugfix-2.0.x branch and unmodified configs from there, same problem.
See attached image.
[image]

*I tried to download latest "1.1.x" branch to test, but I can't compile it due to [error](https://pastebin.com/d45EQhzD)

Both 1.1.9 and 2.0 configs have **#define ANET_FULL_GRAPHICS_LCD**
I checked **\Marlin-1.1.9\Marlin\pins_ANET_10.h** and **\Marlin-2.0\Marlin\src\pins\sanguino\pins_ANET_10.h** and they have same pin numbers. 
So problem is somewhere else.
Any ideas where it can be?

Vertabreak, 2019-12-20:
needs delays i use the following turn them up till the issues stops.
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define ST7920_DELAY_1 DELAY_NS(200)
#define ST7920_DELAY_2 DELAY_NS(200)
#define ST7920_DELAY_3 DELAY_NS(200)

Uinden, 2019-12-20:
@Vertabreak 
Wow, it helped. 
Thank you!
I commented `#define ANET_FULL_GRAPHICS_LCD` and replaced `//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`
To
```
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define ST7920_DELAY_1 DELAY_NS(150)
#define ST7920_DELAY_2 DELAY_NS(150)
#define ST7920_DELAY_3 DELAY_NS(150)
```
And now I have working screen. (100ns still had problem so I used 150ns)

If someone will change configs for Anet A6 in this repo, please also uncomment speaker. It has been disabled since 1.1.9 and Anet A6 has speaker. I use
```
#define SPEAKER
#define LCD_FEEDBACK_FREQUENCY_DURATION_MS 40
#define LCD_FEEDBACK_FREQUENCY_HZ 200
```
And it brings back menu beeps and plays G-code music without any problem.

Vertabreak, 2019-12-20:
i can PR that.

Uinden, 2019-12-20:
@Vertabreak 
I'm just starting to configure everything and I think I see another problem in 2.0 config.
There is
```
#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)    // Y point for Z homing when homing all axes (G28).

  //Anet A6 with new X-Axis
  //#define Z_SAFE_HOMING_X_POINT 113    // X point for Z homing when homing all axes (G28).
  //#define Z_SAFE_HOMING_Y_POINT 112    // Y point for Z homing when homing all axes (G28).

  //Anet A6 with new X-Axis and defined X_HOME_POS -7, Y_HOME_POS -6
  //#define Z_SAFE_HOMING_X_POINT 107    // X point for Z homing when homing all axes (G28).
  //#define Z_SAFE_HOMING_Y_POINT 107    // Y point for Z homing when homing all axes (G28).

#endif
```
So its smashing nozzle in center of bed. Not sure why its under "safe" mark.
Maybe it got accidentally ported from delta printers or something.
After commenting it printer **safely** homes outside of bed.
[image]
You may want to disable it/modify it. 
But I'm not sure. Maybe there is good reason to home in center of bed.

Vertabreak, 2019-12-20:
z safe home is required for probes.

Vertabreak, 2019-12-20:
added it to the PR.

Vertabreak, 2019-12-21:
if you encounter other problems let me know i can add more adjustments should they be required.

Uinden, 2019-12-21:
Configured everything I need, printed some models for testing and did not got any issues related to config.
So its solved.

closed 2019-12-21 by Uinden

kdupke, 2020-01-04:
This helped me. I had the same mangled display when updating to 2.0.

May I propose to add the following lines around line 1945 in Configuration.h?

`
// if your display mangles the content, you might have a slow (maybe older) display. Uncomment the following might help you in this case
#define SLOW_DISPLAY

#if ENABLED(SLOW_DISPLAY)
  #define ST7920_DELAY_1 DELAY_NS(150)
  #define ST7920_DELAY_2 DELAY_NS(150)
  #define ST7920_DELAY_3 DELAY_NS(150)
#endif`

Best regards,
kai

italocjs, 2020-01-16:
I have experienced this for the first time in a ramps1.4, same lcd.  which is extremely weird because i have burned literally dozens of boards using the same program. (bugfix 2.0.x from around may of last year)   Maybe its something related to the u8glib?

italocjs, 2020-01-16:
Nothing changed (as far as i know) related to the environment ive been using (win10, 1908 / no antivirus or stuff/ arduino 1.8.10 / u8glib 1.19.1)

slipx06, 2020-01-18:
> needs delays i use the following turn them up till the issues stops, basically it cant keep up.
> #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
> #define ST7920_DELAY_1 DELAY_NS(200)
> #define ST7920_DELAY_2 DELAY_NS(200)
> #define ST7920_DELAY_3 DELAY_NS(200)

This worked for me. Thank you

adam57jerome, 2020-04-29:
super merci pour toutes les infos