MarlinFirmware/Marlin #16284: [BUG] TMC homing current stays while printing

Lyr3x, 2019-12-21:
### Bug Description
I've set a specific homing current for my TMC5160 to make the sensorless homing working more reliable. However the set homing current is still used when starting the print, which may cause issues due to the lower current. 

### My Configurations
Board: SKR Pro 1.1
Marlin: bugfix-2.0
Driver: TMC5160
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3991313/Configuration.zip)
### Steps to Reproduce

1. Set homing current to e.g. 500mA for X and Y and X/Y_current to 950mA 
2. Start a print
3. Check motor currents when the print started

**Expected behavior:**
Motor current is at 500mA while homing and at 950mA as soon as the homing is done

**Actual behavior:**
Motor current stays at 500mA

boelle, 2020-01-07:
@Lyr3x is this still a problem?

Lyr3x, 2020-01-07:
I'll update to the latest state tonight and will report then.

Lyr3x, 2020-01-13:
Seems to be working fine again.

closed 2020-01-13 by Lyr3x