MarlinFirmware/Marlin #16285: [BUG] DAC steppers and 32 bits MCU

dagorel, 2019-12-21:
Hi all,

### Bug Description

There are issues when using both DAC steppers feature and any 32 bits MCU.

### My Configurations

You can find an example configuration in the attached archive file : [dac+32bits.zip](https://github.com/MarlinFirmware/Marlin/files/3991337/dac%2B32bits.zip).

### Steps to Reproduce

I just changed the board to _BOARD_SMOOTHIEBOARD_ and add stuff to enable DAC features in the smoothisboard pins files. Then i tried to compile: It fails as three avr arduino core macros are not defined for lpc1769: **word**, **highByte**, **lowByte**.

I naively tried to define them in _dac_mcp4728.cpp_:
`
#ifdef MCU_LPC1769
  uint16_t makeWord(byte h, byte l);

  #define word(...) makeWord(__VA_ARGS__)
  #define lowByte(w) ((uint8_t) ((w) & 0xff))
  #define highByte(w) ((uint8_t) ((w) >> 8))
#endif
`

Then the compilation succeeds but it doesn't work as expected as there absolutely no voltage on the DAC (MCP4728) outputs.

**Expected behavior:** DAC steppers feature must work on any 32 bits MCUs.

**Actual behavior:** Compilation failure or no voltage on DAC outputs.

#### Additional Information

I am not sure of what is a _word_ on 32 bits MCUs compared to 8 bits ones.
I am aware there is a built-in DAC on the lpc1769 but i don't use if for now (use of a separate MCP4728).

Feel free to ask for more information. I would be glad to answer.

boelle, 2020-01-07:
@dagorel is this still a problem?

dagorel, 2020-01-07:
Hi @boelle ,

Yes it is. I don't find any fix for now.

dagorel, 2020-01-07:
Hi again @boelle ,

It seems there is an issue in the arduino fmk for lpc176x regarding the missing macros (discussed with @p3p ). I will send a PR to the appropriate github repo.

Fixing this part doesn' fix the whole stuff. As said in the first message there is no voltage on mcp chip outputs... still investigating.

dagorel, 2020-01-14:
Hi @boelle,

Looks like the main issue is on choosing the I2C master address at the platform level.
I already managed it but there is still some issues.
I don't think it can be specifically related to the Marlin repo for now. That's why i close the issue for now.

Regards,
Orel

closed 2020-01-14 by dagorel