MarlinFirmware/Marlin #16289: [FR] Can you add exemple files for JGaurora Magic Maker 3d Printer

sfm67, 2019-12-21:
### Description

Can you add exemple files for JGaurora Magic Maker since jgaurora has publish source code.

thanks

DerAndere1, 2019-12-21:
My 7zip cannot extract JGaurora's firmware archive folder for the JGaurora Magic from http://www.jgaurora3d.com/list_21.aspx?page=1. 
@sfm67: If you can drag-and-drop a functional zip archive containing JGaurora's configuration.h and configuration_adv.h file in a comment here, I could prepare a pull request.

Vertabreak, 2019-12-21:
just ported it and added to PR.

DerAndere1, 2020-01-10:
@sfm67 You can download latest bugfix-2.0.x. It should contain the example config. If it works for you, please close this issue.

Vertabreak, 2020-01-10:
If you have any issues let me know I'll assist n publish any changes as needed it was a direct port should be fine.

sfm67, 2020-01-14:
hi, i have D/L Marlin 2.01, and only see example files for Jgaurora A1, A5 and A5S.

thanks

closed 2020-01-14 by sfm67