MarlinFirmware/Marlin #16291: [FR] Keep showing LCD_SHOW_E_TOTAL after printjob

Lord-Quake, 2019-12-22:
I like using the option

`// Show the E position (filament used) during printing`
`#define LCD_SHOW_E_TOTAL`

Unfortunately the information on screen disappears when the print is complete.

FR would be to keep the information active on screen until the user makes any kind of input change.

oscar6502, 2020-02-24:
It may be a bug but if you disable this option once the printing is finished it remains on the screen

I use Marlin 2.0.4.3 on the skr pro v1.1 board

// #define PRINTJOB_TIMER_AUTOSTART

thinkyhead, 2020-02-24:
We'll set a flag that E should be displayed instead of just looking at the print job status. If the print job is over and you interact with the LCD we'll clear the flag.

Lord-Quake, 2020-06-25:
I've downloaded todays commit. Unfortunately the "cosmetic" issue remains. After the print is complete the screen resets and you no longer know how much filament was used. Sadly I lack the know-how to make such deep changes in the code.

thinkyhead, 2020-06-25:
> After the print is complete the screen resets

I've added a global state to Marlin, and that would be good for this situation. I can add a "job-done" state, which will be useful for maintaining other things too, like keeping the percentage at 100%, which is currently done by (me) hackishly setting an extra bit on the progress percentage value.