MarlinFirmware/Marlin #16292: [BUG] pt100 not work skr 1.3

userosos, 2019-12-22:
If i set #define TEMP_SENSOR_0 1 to #define TEMP_SENSOR_0 20 and build it for LPC1768. After flash it to skr 1.3 the motherboard not connect to PC or raspberrypi. I can't see the com port.
I use the manual: https://e3d-online.dozuki.com/Wiki/E3D_PT100_Amplifier_Documentation
https://dropmefiles.com/m5JjY - The full sources

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3992445/Marlin.zip) - The configs files.

AnHardt, 2019-12-22:
You did not follow  https://e3d-online.dozuki.com/Wiki/E3D_PT100_Amplifier_Documentation .

You used one of the standard temperature ADC ports with a 4.7KOrm pullup resistor.
You have to user an ADC pin without a pullup resistor.
For that you have to change the pin-file.

Else Marlin does detect a temperature sensor failure and kills the process before the host connection is established.

ellensp, 2019-12-22:
If you used P0_24_A1  (standard TEMP_0_PIN  on a skr 1.3) the data sheet https://www.nxp.com/docs/en/data-sheet/LPC1769_68_67_66_65_64_63.pdf says that P0[24]  "When configured as a ADC input, digital section of the pad is disabled and the pin is not 5 V tolerant."  Did you send 5v to a non 5v tolerant pin?

userosos, 2019-12-22:
@AnHardt Ok! thx!
@ellensp I not connected pt100 with amplefaire now.
Which port can I use on an skr 1.3 for PT100?

AnHardt, 2019-12-22:
> Which port can I use on an skr 1.3 for PT100?

Non - if you want to be really save. No of the pins is tolerating more than 3.3V (VCC) in ADC mode - neither an LPC nor an a ST.
And all ADC capable - if you are absolute sure your temperatures **!!!never!!!** will be above ~550°C and you are able to make your own table.
> 500 | 3.00
600 | 3.33
700 | 3.63

To be safe you should use a voltage divider, reducing 5V to  3.3V (ideally with a 10 turn Potentiometer 
 of about 5K or 10K ). Only then you can use table 20.

atoomnetmarc, 2019-12-22:
@userosos 
For me removing the onboard pull-up resistor from the SKR board and powering the PT100 amplifier from 3.3V just works. No need to make your own tables.

karabas2011, 2019-12-23:
> For me removing the onboard pull-up resistor from the SKR board and powering the PT100 amplifier from 3.3V just works. No need to make your own tables.

The same for me, but for room temp it reads temp higher for 4-6 degrees then actual. I think it is due to low voltage. Max temp I got was 400-420C ( only all metal hotend)

Bob-the-Kuhn, 2019-12-23:
Lots of reasons the reading can be off by 4-6 degrees.  I wouldn't bother trying to fix that.  Lots of work for no improvement in the print quality.

You need to tune the extruder temperature for every batch of filament you get.  All that matters is using the temperature (as reported by the on-board thermocouple/RTD/...) that works best.  Doesn't matter if the secondary method of reading the extruder temperature says something different.

Bob-the-Kuhn, 2019-12-23:
FYI - I threw away my PT100 amplifier and just did a direct connect with a 1K pullup.  Using a 4.7K pullup lowers the resolution but it's still better than that seen by the standard EPCOs 100k at 200C (and above).

karabas2011, 2019-12-23:
Why it is better than amplifier? Precision resistor is required

Bob-the-Kuhn, 2019-12-24:
What is acceptable precision?  

I maintain that what you really need is:
- repeatability - the ability to get back to the same temperature time after time so that the prints are consistent
- resolution - if the resolution is too course then repeatability is poor and the printer will continually hunt for the desired temperature

When you do your test prints to calibrate a new batch of filament most of the time the temperature increments are 10 degrees C.  When it's all over you'll just put the setpoint at where ever the test prints look best.

A precision resistor is only needed if there is a need to accurately know the temperature.  When melting filament all we need to know is the general temperature so that the tuning process can be started .  The tuning will determine what setpoint to use.

Grogyan, 2019-12-24:
Not correct, as the optimal temperature should not change beyond what the manufacturer specified.
As it is, thermistors generally read 2 - 3 degrees higher than a thermocouple. 
There is a need for a paradigm shift in consumer 3d printers to use robust sensors like what is used in industrial situations. 
High accuracy and reliability, especially with something as critical as the hot end of a 3d printer should not be under estimated. 

Am sorry to hear that you could not not get the PT100 amplifier working. 
There is a similar thread going for using a PT100 with a MAX31865

Bob-the-Kuhn, 2019-12-25:
My position is that, for hobbiest printers, knowledge of the absolute temperature is not nearly as important as finding the indicated temperature at which optimal results are obtained.  

> Not correct, as the optimal temperature should not change beyond what the manufacturer specified.

1. There's a lot of people that think the optimal temperature varies from batch to batch even from the same manufacturer.  They're the ones recommending doing test prints with each new spool.  On the other hand I haven't changed my setpoint for a long time even though I've had multiple spools, multiple colors and multiple manufacturers.  I have to confess that I'm mostly interested in functionality so the variations I get in print quality are acceptable to me.
2. Optimal varies depending on what goals you are aiming for.  In my experience I get stronger parts at higher temperatures but bridging is better at lower temperatures.  
 

> use robust sensors like what is used in industrial situations

Industrial situations are **VERY** interested in being able to reliably reproduce results across multiple machines using the **same** recipe and the **same** batch of raw materials.  In that situation I agree that precise knowledge of the temperature is required.

---

I have no problems hooking up and optimizing a PT100 amplifier.  I just think it's not worth the hassle of the custom wiring.  Take a look at some of the older PT100 issues and you'll see my name on a lot of them.

karabas2011, 2019-12-26:
BTW I was told that PT100 temp offset is constant along the full range. If your room temp is exactly 24C and PT100 reading is 29C you need to subtract 5 degrees in firmware. And 200C reading also will become exact.

Bob-the-Kuhn, 2019-12-28:
It depends on the source of the error.

Most likely it's due to a voltage offset due to current flowing through the ground traces/wiring making the ground voltage seen by the ADC different than what's at the PT100 amplifier.  If that's the case then the offset will most likely be constant.

If the offset is due to the pullup resistor being off then the offset will change with temperature.  A 2% error in the 4400 pullup within the PT100 amplifier would result in about a 5C error at room 20C and about 9C at 220C.

boelle, 2020-01-07:
@userosos is this still a problem?

karabas2011, 2020-01-13:
> It depends on the source of the error.
> 
> Most likely it's due to a voltage offset due to current flowing through the ground traces/wiring making the ground voltage seen by the ADC different than what's at the PT100 amplifier. If that's the case then the offset will most likely be constant.
> 
> If the offset is due to the pullup resistor being off then the offset will change with temperature. A 2% error in the 4400 pullup within the PT100 amplifier would result in about a 5C error at room 20C and about 9C at 220C. I would be really surprised if the PT100's pullup resistor tolerance was worse that 1%. Most likely they used 0.1% or 0.5% parts.
So it is reasonable to have Gcode command to set offset. In most cases pt100 reading is about 4-6С higher at room temperature. Subtracting it make higher reading more precious.

boelle, 2020-01-18:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-18 by boelle