MarlinFirmware/Marlin #16294: [BUG] Errors when compiling for anet a8 platformio

albertoxamin, 2019-12-22:
### Bug Description

I tried to build for `sanguino_atmega1284p`

and got this

```bash

> Executing task: platformio run --environment sanguino_atmega1284p <

Processing sanguino_atmega1284p (platform: atmelavr; board: sanguino_atmega1284p; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/sanguino_atmega1284p.html
PLATFORM: Atmel AVR 2.0.0 > Sanguino ATmega1284p (16MHz)
HARDWARE: ATMEGA1284P 16MHz, 16KB RAM, 127KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduino-avr 5.0.0
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 16 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.1
|   |-- <SPI> 1.0
|   |-- <SoftwareSerial> 1.0
|-- <Adafruit NeoPixel> 1.2.5
|-- <Adafruit MAX31865 library> 1.0.2
|   |-- <SPI> 1.0
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <TMC26XStepper>
|   |-- <SPI> 1.0
|-- <Wire> 1.0
|-- <SPI> 1.0
Building in release mode
Compiling .pio/build/sanguino_atmega1284p/src/src/HAL/HAL_AVR/HAL.cpp.o
Compiling .pio/build/sanguino_atmega1284p/src/src/HAL/HAL_AVR/HAL_SPI.cpp.o
In file included from Marlin/src/HAL/HAL_AVR/../../inc/../pins/pins.h:71:0,
                 from Marlin/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:32,
                 from Marlin/src/HAL/HAL_AVR/HAL.cpp:25:
Marlin/src/HAL/HAL_AVR/../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^
In file included from Marlin/src/HAL/HAL_AVR/../../inc/../pins/pins.h:71:0,
                 from Marlin/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:32,
                 from Marlin/src/HAL/HAL_AVR/HAL_SPI.cpp:34:
Marlin/src/HAL/HAL_AVR/../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^
Compiling .pio/build/sanguino_atmega1284p/src/src/HAL/HAL_AVR/MarlinSerial.cpp.o
Compiling .pio/build/sanguino_atmega1284p/src/src/HAL/HAL_AVR/Servo.cpp.o
In file included from Marlin/src/HAL/HAL_AVR/../../inc/../pins/pins.h:71:0,
                 from Marlin/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:32,
                 from Marlin/src/HAL/HAL_AVR/MarlinSerial.cpp:39:
Marlin/src/HAL/HAL_AVR/../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^
In file included from Marlin/src/HAL/HAL_AVR/../../inc/../pins/pins.h:71:0,
                 from Marlin/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:32,
                 from Marlin/src/HAL/HAL_AVR/Servo.cpp:56:
Marlin/src/HAL/HAL_AVR/../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^
*** [.pio/build/sanguino_atmega1284p/src/src/HAL/HAL_AVR/HAL.cpp.o] Error 1
*** [.pio/build/sanguino_atmega1284p/src/src/HAL/HAL_AVR/HAL_SPI.cpp.o] Error 1
*** [.pio/build/sanguino_atmega1284p/src/src/HAL/HAL_AVR/MarlinSerial.cpp.o] Error 1
*** [.pio/build/sanguino_atmega1284p/src/src/HAL/HAL_AVR/Servo.cpp.o] Error 1
============================================================================= [FAILED] Took 6.64 seconds =============================================================================

Environment                   Status    Duration
----------------------------  --------  ------------
megaatmega2560                IGNORED
megaatmega1280                IGNORED
rambo                         IGNORED
FYSETC_F6_13                  IGNORED
sanguino_atmega644p           IGNORED
sanguino_atmega1284p          FAILED    00:00:06.637
melzi                         IGNORED
melzi_optiboot                IGNORED
at90usb1286_cdc               IGNORED
at90usb1286_dfu               IGNORED
DUE                           IGNORED
DUE_USB                       IGNORED
DUE_debug                     IGNORED
LPC1768                       IGNORED
LPC1769                       IGNORED
STM32F103RC                   IGNORED
STM32F103RC_fysetc            IGNORED
STM32F103RC_bigtree           IGNORED
STM32F103RC_bigtree_USB       IGNORED
STM32F103RC_bigtree_512K      IGNORED
STM32F103RC_bigtree_512K_USB  IGNORED
STM32F103RE                   IGNORED
STM32F103RE_bigtree           IGNORED
STM32F103RE_bigtree_USB       IGNORED
STM32F4                       IGNORED
STM32F7                       IGNORED
ARMED                         IGNORED
STM32F103VE_longer            IGNORED
mks_robin                     IGNORED
mks_robin_lite                IGNORED
mks_robin_mini                IGNORED
mks_robin_nano                IGNORED
jgaurora_a5s_a1               IGNORED
STM32F103CB_malyan            IGNORED
chitu_f103                    IGNORED
FYSETC_S6                     IGNORED
STM32F407VE_black             IGNORED
BIGTREE_SKR_PRO               IGNORED
BIGTREE_BTT002                IGNORED
teensy31                      IGNORED
teensy35                      IGNORED
esp32                         IGNORED
linux_native                  IGNORED
SAMD51_grandcentral_m4        IGNORED
include_tree                  IGNORED
======================================================================== 1 failed, 0 succeeded in 00:00:06.637 ========================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

### My Configurations

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3992701/Archive.zip)

ellensp, 2019-12-23:
You have set #define MOTHERBOARD BOARD_RAMPS_14_EFB   this is a mega2560 not a sanguino_atmega1284p  The real A8 configuration.h files has #define MOTHERBOARD BOARD_ANET_10

Vertabreak, 2019-12-23:
is this something that should be adjusted in the example config? i can suggest the change should it be required.

ellensp, 2019-12-23:
It is in the example A8 config correctly https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/config/examples/Anet/A8/Configuration.h#L133

Vertabreak, 2019-12-24:
try the melzy environment in platformio to compile for the atmega1284p.

albertoxamin, 2019-12-26:
Setting the env `melzi` results in a successful compilation, also changing the config to https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/config/examples/Anet/A8/Configuration.h#L133 worked!

Thanks guys!

closed 2019-12-26 by albertoxamin