MarlinFirmware/Marlin #16298: [BUG] ( Timing For Rotary Encoder Not Independent of Processor Speed on LPC1768 )

borland1, 2019-12-22:
### Bug Description

Graphics Display's rotary encoder works great with Configuration files when compiled for 8-bit processor Atmega 2560, however with same Configuration files, when compiled for 32-bit processor LPC1768, rotary encoder sometimes causes pixel corruption on display.  

### My Configuration 

Timing for Rotary Encoder seems off, because this can be corrected with a modification to Configuration_adv.h file, namely changing ENCODER_RATE_MULTIPLIER rates (steps/sec) to half of default.
In Configuration_adv.h file, Under "Additional Features"... line 903...

#define ENCODER_RATE_MULTIPLIER
#if ENABLED(ENCODER_RATE_MULTIPLIER)
  #define ENCODER_10X_STEPS_PER_SEC   15  // (steps/s) Encoder rate for 10x speed, default 30
  #define ENCODER_100X_STEPS_PER_SEC  40  // (steps/s) Encoder rate for 100x speed, default 80
#endif

#### Additional Information

[ConfigurationFiles.zip](https://github.com/MarlinFirmware/Marlin/files/3992908/ConfigurationFiles.zip)

dahuafschmied, 2019-12-25:
@broland1 i can confirm on the skr1.3 your fix is a solution for the encoder problem. thank you!

borland1, 2019-12-25:
Glad you confirmed the timing problem also exists with skr1.3 board, however my solution is only a temporary fix and workaround to an existing bug with subject processor (LPC1768).

Should this timing bug be fixed at later date, the workaround may require a reset to default values.

boelle, 2020-01-07:
@borland1 is this still an issue?

borland1, 2020-01-07:
Yes, still an issue.  For LPC1768 and all other processors, variable ENCODER_10X_STEPS_PER_SEC should be the same setting (steps/second) for same rotary encoder hardware and independent of processor speed.

boelle, 2020-01-24:
how often do you update? bugfix is updated almost daily

borland1, 2020-01-24:
Only when I see something that may improve upon my hardware build.  What has changed with bugfix that is applicable here?

boelle, 2020-01-24:
i dont know, i'm not a coder or maintainer

i just asked since a lot have happended during the last 17 days and it could have been fixed

but personally i dont have an issue with the encoder ony my displays, i have a full graph one and a char based one (like on prusa)

boelle, 2020-01-24:
but for those that have a problem will it be fixed for all with your config changes in first post?
it seems so... and in that case i will change title to say we also have a fix here

borland1, 2020-01-24:
OK, however if related timing is later modified by some other timing related 2.0.x.bugfix, the above workaround, of changing those parameters, will likely be unsatisfactory.

It seems like this may be a processor Timer/Counter calibration (with processor speed) problem. There are also variations of LPC1768 with differing processor clock speeds up to 100MHz.  And other similar ARM Cortex-M3 processors (LPC1769) with clock speed up to 120MHz.  These may be affected as well.

boelle, 2020-01-30:
a lot of updates and fixing has happend in the last week, is the problem still there?

borland1, 2020-01-30:
Tell me what specifically has changed which might affect the this timing bug.  Too much activity here, so I do not follow every PR due to time constraints.
If code was much smaller, I would take the time to debug the problem.  I scanned some of the code for the cause, but was quickly overwhelmed.

boelle, 2020-01-30:
> , I do not follow every PR.

nor do i

> Too much activity here,

agreed hence i do not follow everything, maintaining the issue list is more than enough

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle