MarlinFirmware/Marlin #16299: [BUG] TMC2208 RXTX_pin not declared when SW_CAPABLE_PLATFORM not defined

mikeshub, 2019-12-22:
Why did TMCStepper.h get changed so that it include software serial even if the drivers don't use serial or if software serial is needed or not? I like endstop interrupts on my printer. To get it to compile I commented out lines 59-61 from SanityCheck.h and then I commented out TMCStepper.h line 19. 

Onto the bug I am reporting. TMCStepper.h line 987 `const uint16_t RXTX_pin = 0; // Half duplex` should be below the #endif otherwise it won't compile.

Was:

```
		#if SW_CAPABLE_PLATFORM
			SoftwareSerial * SWSerial = NULL;
			const uint16_t RXTX_pin = 0; // Half duplex
		#endif
```

Should be:
```

		#if SW_CAPABLE_PLATFORM
			SoftwareSerial * SWSerial = NULL;
		#endif
			const uint16_t RXTX_pin = 0; // Half duplex
```

What's the point of including software serial if there aren't any UART drivers?

teemuatlut, 2019-12-22:
Because an external library doesn't know of your Marlin configuration.

mikeshub, 2019-12-22:
I'm not explaining this correctly. First part is how I found the bug. I am not advocating any changes like that here. It might be worth messing with on the dev or bug fix branch, but it's a pretty low priority. 

As to the bug, I'm not sure there are configurations that are not SW_CAPABLE_PLATFORM and would use Trinamic, but if there are they will not compile with a Trinamic driver SPI or UART. If this preprocessor directive is not met:

```
#if (__cplusplus == 201703L) && defined(__has_include)
	#define SW_CAPABLE_PLATFORM __has_include(<SoftwareSerial.h>)
#else
	#define SW_CAPABLE_PLATFORM defined(__AVR__) || defined(TARGET_LPC1768) || defined(ARDUINO_ARCH_STM32)
#endif
```
and you are using Trinamic drivers, even SPI ones, it won't compile. Which will be an issue on the off chance someone has a configuration that uses Trinamic drivers, but doesn't have software serial capabilities.

The error comes from TMC2208Stepper.cpp line 135 and 144. It should be able to use a hardware serial in a platform without software serial and TMC2208s. My previous comment is about what prevents that from happening.

mikeshub, 2019-12-22:
Oh I just got what you meant! I'll report this over at trinamic.

teemuatlut, 2019-12-23:
There's no bug and reporting to Trinamic won't achieve anything.

The lines 135 and 144 are compiled only for AVR which *IS* a SW capable platform and thus will have the member variable declared.
You can test by compiling the Simple example for DUE, which is one platform that does not have a SoftwareSerial library included.
The same code is compiled, doesn't matter if you're using SPI or UART.

boelle, 2020-01-07:
closing until configs are provided

closed 2020-01-07 by boelle

mikeshub, 2020-01-07:
Closing this was the right action. Sorry for the slow roll on this I've been super busy. I intend on just making the PR myself once I can finally get around to it. It is a pretty simple fix of just moving the software serial include to the drivers that use software serial.

And yes it IS a bug because it is 1. not needed and 2. breaks endstop interrupts. Whether the platform has software serial or not is irrelevant to the fact that it breaks endstop interrupts on AVR platforms. Doubly so when it does this and you aren't even using a UART driver.

mikeshub, 2020-01-07:
Just in case you are interested here are the zipped files per your request. 
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4028604/Configs.zip)