MarlinFirmware/Marlin #16302: [BUG](bugfix-2.0.x) (SKR Mini 1.2) Thermal Runaway Protection is not kicking in when unplugging the thermistor

Seref, 2019-12-23:
When heating up the bed and the hotend to 60°C and 250°C and unplugging it after they reached said temperature the current state is stuck.
With state I refer to state of the mosfets, if it was on while unplugging it it will stay on.

Added to that MIN_TEMP is also not working when unplugging the thermistor in the beginning eventhough the current temperature has a negative value. (When unplugging the thermistor while the system heats up the temperature is stuck).

I'm using Pronterface to control the 3D printer and set the values.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3993710/Configuration.zip)

Seref, 2019-12-24:
Okay I tried to compare some version (either bugfix or 2.0.x) but it really seems like thermal protection is broken.

Seref, 2019-12-31:
I found a partial solution, disabling PID control for the heatbed atleast shutsdown when the board crashes, still no error is thrown and nothing else happens it just crashes a after a while, not even kill() or anything compareable is called.

Seref, 2019-12-31:
Okay after some more hours of testing I think I'm able to pinpoint it a bit more, it seems like the serial connection gets worse when using composite USB aka **STM32F103RC_bigtree_USB** or **STM32F103RC_bigtree_512K_USB**.

So for everyone who also got this problem with this board:
Just use the non USB version (**STM32F103RC_bigtree** & **STM32F103RC_bigtree_512K**), additionally now when using PID_TEMP for the heated bed it also shuts down correctly.

There still seems to be a bug with the serial connection because the error is only partially printed:

> **E**
[image]

I think this is from the error message because it is the only thing that gets printed right after I unpluged the thermistor, the good thing now is that if the heated bed or the hotend gets MIN_TEMP triggered it won't be stuck in the current powering state it really shuts down.

This isn't a really fix as it is a workaround for the problem, and the problem itself is has more to do with the USB connection, atleast that is my current assumption.

boelle, 2020-01-07:
@Seref still an issue?

boelle, 2020-01-14:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-14 by boelle

Seref, 2020-01-15:
I didn't really solve it, and as mentioned before I just turned off the composite USB feature, which fixed kind of fixed it.