MarlinFirmware/Marlin #16303: [BUG] Extruder that revolves still only one direction

realbehype, 2019-12-23:
Hey, you met Marlin to SKR 1.3 with an extruder that revolves still only one direction? 😀 I no longer know what to do with it, whether I put any driver, any cable or motor on the extruder, so whether I turn the extruder forward or backward, it still rotates in one direction only.

ellensp, 2019-12-23:
no configuration files...  no evidence...   No one else has this issue with this controller...  most likely a board fault specific to your board.

realbehype, 2019-12-23:
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3993880/configuration.zip)

I'm sorry, I attach files.

ellensp, 2019-12-23:
Did you edit the file pins_BTT_SKR_V1_3.h or pins_BTT_SKR.h at all?
Have you tried swapping E0 amd E1 pins over to see if the fault moves?

realbehype, 2019-12-23:
Yes, everything has been tested :/

ellensp, 2019-12-23:
That didn't answer either of the last two questions...

Bob-the-Kuhn, 2019-12-23:
In my experience rotation in only one direction can be caused by any of the following:
- Bad cable
- endstop active (E0 hardly ever has an endstop so this is very unlikely)
- Current set too low
- Microsteps set too high
- Acceleration too high
- stepper bad
- driver module bad
- motherboard bad

If you get to the point where you think the motherboard is bad then you'll need a pins file with E0 and E1 swapped.    **pins_BTT_SKR_V1_3.h** has the E1 socket definitions added.   **pins_BTT_SKR_V1_3.h swapped** has E0 and E1 swapped.  If you use  **pins_BTT_SKR_V1_3.h swapped** you'll need to rename it to **pins_BTT_SKR_V1_3.h** before use.
[pins_BTT_SKR_V1_3.zip](https://github.com/MarlinFirmware/Marlin/files/3994037/pins_BTT_SKR_V1_3.zip)

boelle, 2020-01-07:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

op never answered questions posted

closed 2020-01-07 by boelle