MarlinFirmware/Marlin #16304: SKR 1.1 + LCD REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER = screen garbage

TriggerMeister, 2019-12-23:
[Configuration_current.zip](https://github.com/MarlinFirmware/Marlin/files/3994890/Configuration_current.zip)
[April2019working_Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3994954/April2019working_Configuration.zip)

### Bug Description

The problem is that I'm getting "garbage" on the LCD screen.  This was working in a prior Marlin 2.0 version, but something changed in Marlin or the way Marlin is compiled?

**Hardware:**
- Marlin 
- SKR 1.1
- Z-endstop inductive / BLTouch
- REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

**Timeline:**
**Prior to the April 2019** I was running Marlin 2.0 with inductive probe on the HEVO printer, bilinear bed leveling.  LCD was working fine.
**In April 2019** I switched to to a BL touch clone, recompiled Marlin with the necessary configurations.  LCD was working fine
**In November 2019** I decided to remove the BL touch and go back to the inductive probe.  When I tried to recompile i was presented with re-compile problems because of a PlatformIO upgrade.  Hence, I grabbed the latest copy of Marlin 2.0, updated the stock config to match my hardware, etc.  I was able to compile, but was presented a issue with Bilinear bed leveling (stopped after probing 6 out of 9 points... was listed as a open issue at the time).  Switched to another working bed leveling method that worked.  Recompiled again.  After running for several minutes the screen garbage appears and it's seems to be confined to the top section that shows the heater temp appears.  To note, I have another printer with the same type of hardware setup, running the same marlin version with a re-arm board and it does not have this screen garbage issues.  As a reminder this was working fine in a previous Marlin version on the printer in question.  I'm also building a new printer with a new SKR 1.1. and same LCD that is also experiencing this issue, just so we can rule out the "hardware".  I followed the recommendations to add/increase the delay, which seems to only delay the issue from appearing.  

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce
- Do a fresh git pull for Marlin
- compile with attached configuration.h
- install on SKR 1.1 board and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

**Expected behavior:** [What you expect to happen]
Screen should work like it did in a earlier version. 

**Actual behavior:** [What actually happens]
Garbage appears on the LCD screen after a couple minutes of running. 

Delay of 200:
[image]

Worst scenario:
[image]

[image]

P.S. the pictures are from the new printer I'm building since it was more convenient to take them, but the LDC issue is identical on both printers.

#### Additional Information

* I can continue to play with the delays, but so far I'm at 450 and it still appears. The real question is what changed that caused the LCD to no longer work out-of-the-box?

TriggerMeister, 2019-12-25:
Merry Christmas!

**Here is a X-mas update:**  To test, I removed EXP2 connection from the board & LCD and the screen is working fine with a fresh pull, fresh compile, no added delays, etc.  The screen has been stable for almost 2 hours now with no garbage **(OF COURSE MY ROTARY ENCODER IS NOT WORKING WITH EXP2 DISCONNECTED)**. So the issue seems to be connected to EXP2, like possibly a wrong pin being used for SD card or rotary encoder.  I looked at the pin config in  pins_BTT_SKR_V1_1.h and it appears that there have been some significant changes from the last working version.  I'm guessing that there is some conflict in the pin output.  It also looks like the code was split to make the SKR code more reusable.  

Below is the pin file from when it was working.  Back then it was a single file and it was named BIQU_SKR_V1.1.  Can someone take a look if the pins are now different?  

`/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

#ifndef TARGET_LPC1768
  #error "Oops!  Make sure you have the LPC1768 environment selected in your IDE."
#endif

#ifndef BOARD_NAME
  #define BOARD_NAME "BIQU SKR V1.1"
#endif

//
// Limit Switches
//
#define X_MIN_PIN          P1_29
#define X_MAX_PIN          P1_28
#define Y_MIN_PIN          P1_27
#define Y_MAX_PIN          P1_26
#define Z_MIN_PIN          P1_25
#define Z_MAX_PIN          P1_24

//
// Steppers
//
#define X_STEP_PIN         P0_04
#define X_DIR_PIN          P0_05
#define X_ENABLE_PIN       P4_28

#define Y_STEP_PIN         P2_01
#define Y_DIR_PIN          P2_02
#define Y_ENABLE_PIN       P2_00

#define Z_STEP_PIN         P0_20
#define Z_DIR_PIN          P0_21
#define Z_ENABLE_PIN       P0_19

#define E0_STEP_PIN        P0_11
#define E0_DIR_PIN         P2_13
#define E0_ENABLE_PIN      P2_12

#define E1_STEP_PIN        P0_01
#define E1_DIR_PIN         P0_00
#define E1_ENABLE_PIN      P0_10

//
[... 184 lines cut ...]
      #define E0_CS_PIN     P1_31
      // We use LCD_SDSS pin for E1
      #undef  LCD_SDSS
      #define LCD_SDSS      -1
      #define E1_CS_PIN     P1_23
    #endif

    // Example 3: Use the driver enable pins for chip-select.
    //            Commands must not be sent to the drivers when enabled. So certain
    //            advanced features (like driver monitoring) will not be available.
    //#define SKR_USE_ENABLE_CS
    #if ENABLED(SKR_USE_ENABLE_FOR_CS)
      #define X_CS_PIN      X_ENABLE_PIN
      #define Y_CS_PIN      Y_ENABLE_PIN
      #define Z_CS_PIN      Z_ENABLE_PIN
      #define E0_CS_PIN     E0_ENABLE_PIN
      #define E1_CS_PIN     E1_ENABLE_PIN
    #endif

  #endif // SOFTWARE_DRIVER_ENABLE

#endif
`

Thanks!

aurion55, 2019-12-26:
I got the same problem as above, I was using Marlin Bugfix_2.X for ages and decide to update to Marlin 2.0, I'm running a MKS Gen 1.4 board display is like above, expectt mind start straight away from boot up.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4000934/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4000935/Configuration_adv.zip)

TriggerMeister, 2019-12-26:
I spoke too soon.  Even with only the EXT1 line connected I ended up receiving garbage on the screen later on in the day.  Still not sure why this happening, but I found certain references:

/**
 * LCD / Controller
 *
 * As of 20 JAN 2019 only the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER display has
 * been tested with these settings. It can be connected to the SKR using standard cables
 * via the EXP1 and EXP2 ports. Other displays may need a custom cable and/or changes to
 * the pins defined below.
 *
 * The SD card on the LCD controller uses the same SPI signals as the LCD, resulting in
 * garbage/lines on the LCD display during SD card access. The LCD code mitigates this
 * by redrawing the screen after SD card accesses.
 */

It also appears that certain pins are being reused for special purposes and folks reference additional noise on the pins:

  // To minimize pin usage use the same clock pin as the display/SD card reader. (May generate LCD noise.)
  #define TMC_SW_SCK        P0_15

So while I was smitten with the board for a while it looks like it is being hacked up to make it something that it's not.

TriggerMeister, 2019-12-27:
Update:  I was able to make the LCD work properly with just the regular USB power supply (changed the jumper).  Anyone trying to duplicate this needs to use the internal power supply jumper setting.  It appears that for whatever reason the 5V (I measured 4.8V actually) ended up being noisier post Marlin update.  I know it will probably be dismissed as a hardware issue, but the board was running fine for long time with the same LCD.  To add to the coincidences, I have another SKR v 1.1 board with the same LCD's that has the same problem.  The LCD's have been verified on a Re-arm board and are working fine on it with the same Marlin version.

My solution to resolving the problem is to add a 100uF capacitor into the ribbon cable.  This resolved the garbled screen issue.  I still have to apply the same solution to the other board.

[image]
[image]
[image]
[image]

TriggerMeister, 2019-12-27:
Added a capacitor to the other printer today with the same setup.  No screen garbage so far (yeah!) on either printer.  I'm not sure why an updated/recompiled marlin after a PlatformIO update would cause more "noise" on the LCD. In a FB group someone suggested it may be caused by Intellisense and the way Marlin is compiled now after the update, which would make sense as I was not able to compile a working old version even after fixing the "upload_extra_script.py" of my previously working Marlin version.  

[image]
[image]

mtraven, 2019-12-29:
Just wanted to add:  I had the same issue when upgrading to 2.0, never had the issue before.  Mine was able to be resolved in software by adding the following lines to config.h:
#define ST7920_DELAY_1 DELAY_NS(0)
#define ST7920_DELAY_2 DELAY_NS(250)
#define ST7920_DELAY_3 DELAY_NS(250)

This seems to be a fix from older version of marlin, but I never had any problems before 2.0

also, my screen did not look quite as bad as yours, so maybe there is a hardware variable to it.

TriggerMeister, 2019-12-29:
One was worse than the other, but the capacitor solved it on both.  The delay occasionally helped, but the longer the printer ran the more likely it was going to come back.  Maybe the power converter to 5V is not as good as it is on other boards (e g. Re-arm).  Just weird that it started after a upgrade....

mtraven, 2019-12-29:
I am having other issues with 2.0 that led me to roll back to bugfix1.1.9.  Interestingly, the screen problem came back.  I guess I never flashed bugfix1.1.9, so I think the prev version I was running was just marlin 1.1.9.  Meaning the problem (for me) started at bugfix1.1.9.

The cap is a clever fix, but what exactly is it doing?  extending those delays in hardware?
also, what happens if they fix this is software, you would probably have to remove the caps, no?

it seems like, even though you have solved your problem, you still want to know what the root cause is?  I am in the same boat, feels like a band aid at this point, really would like to know exactly what changed.

TriggerMeister, 2019-12-30:
I do want to know what changed.  Frankly I would have been better off buying the TFT instead of wasting days on this.

Here is what I know:
1. I can't compile any old GitHub pulls and make it work like it used to.
2. PlatformIO changed and that forced me to get a new pull, otherwise I had compile errors.
3. If you use the USB power selection the LCD works fine.  
4. Adding a cap reduces the noise on the LCD power line and fixes the issue.

I KNOW it worked prior to this upgrade.  I don't think Marlin changed something, but there is a dependency that changed and Marlin needs to ensure that dependency is met.  I mean that the Platform IO is compiling differently and while it's fine for other boards the SKR seems to puke on the order possibly or how it is compiled.  This may throw off the timing and the LCD.... Just guessing.

No, I will not remove the cap.  There is no harm in keeping it even if Marlin is patched.  It's cheap insurance.

Kaiwol1990, 2020-01-06:
I had the same problem or at least a similar one. Mine display just worked fine if I unplugged it for a short time after marlin booted completely. For me the fix was to add the following lines to Configuration.h.

`#define ST7920_DELAY_1 DELAY_NS(0) `
`#define ST7920_DELAY_2 DELAY_NS(0)  `
`#define ST7920_DELAY_3 DELAY_NS(472)`

The 472 are from the [RepRap page](https://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller). There is noted a 63 which is for a 16 MHz MCU. Since the SKR 1.3 (I'm using) has 120 MHz, I scaled it up accordingly. This is due to a slow LCD as the mentioned on the page. 

I hope this helps some people. It may be a suitable fix to add a flag to the Configuration.h and then calculate the delay based on the MCU clock.   Something like:
'#define SLOW_LCD'

TriggerMeister, 2020-01-07:
> I had the same problem or at least a similar one. Mine display just worked fine if I unplugged it for a short time after marlin booted completely. For me the fix was to add the following lines to [...]

I think I went all the way up to 450, but then it ended coming back hours later.  So maybe I just stopped short???  I do believe that there is something with the power applied and the clock speed, as the displays worked fine on USB power.

boelle, 2020-01-14:
@TriggerMeister so this one is not solved yet? i saw some hardware mods and wonder if its a hardware problem rather than a marlin one

TriggerMeister, 2020-01-14:
I don't have more time to burn on this, but I think is very weird that so many people experienced this issue after an upgrade.  I guess nobody will try to figure out why this happened or try to trace it down.  I think it's tied to a dependency and the issue may not be directly in Marlin.  Thanks.

italocjs, 2020-01-16:
I can confirm that i have experienced the same issue, but on another board (ramps 1.4), with the same firmware i have burned dozens of times, so i guess it might be some update to the u8glib library that messed something up.  (arduino is the same, 1.8.10)

boelle, 2020-01-21:
what if vscode+platform.io extension is used?

arduino ide is filled with all sorts of options for errors

italocjs, 2020-01-21:
@boelle i'll compile it with vscode to see if it persists without adding the delays. its a good guess

boelle, 2020-01-21:
i also had all kind of problems with arduino ide, then i tried vscode and i have never looked back

SajjadShooshtari, 2020-01-31:
Hi
I changed the following data and my problem was resolved.
file:
Marlin-2.0.x\Marlin-2.0.x\Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.h
line 44,45,46 :

#define ST7920_DELAY_1 DELAY_NS(0)
#define ST7920_DELAY_2 DELAY_NS(250)
#define ST7920_DELAY_3 DELAY_NS(250)

Good luck

boelle, 2020-06-23:
@TriggerMeister 

Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

sjasonsmith, 2020-07-04:
Given the change that has occurred in Marlin since the last time this was active, I think it would be irresponsible to blindly apply new timings without having the hardware to test it.

Certainly many people are using these screens with new SKR (1.3 + 1.4) boards. If there is still a problem unique to SKR 1.1 boards, we'll need someone with the hardware to volunteer a bit of time to test a fix.

Given the silence and age of this issue, I'm going to close the issue. If you are still having to employ workarounds for this, please reply and we can re-open the issue and figure out an appropriate change to fix it.

closed 2020-07-04 by sjasonsmith