MarlinFirmware/Marlin #16306: [BUG] Not working RESTORE_LEVELING_AFTER_G28

bigtreeteche3, 2019-12-23:
After aktivating:  RESTORE_LEVELING_AFTER_G28
It does not restore the previous leveling stored in the eeprom. You have first to go into the menu and reset bed level manualy to on because it will set itself always back to off. After that it will restore the previous leveling after G28.

Vertabreak, 2019-12-23:
when using other bed leveling that isn't UBL you need to load the mesh before starting the print.

bigtreeteche3, 2019-12-23:
just by adding m420 s to the gcode?

thinkyhead, 2019-12-24:
The `RESTORE_LEVELING_AFTER_G28` option causes `G28` to enable leveling if it was on before `G28`. It doesn't enable leveling if it was off before `G28`.

thinkyhead, 2019-12-24:
> just by adding `M420 S` to the gcode?

That will enable leveling using the mesh currently stored in RAM.

closed 2019-12-24 by thinkyhead