MarlinFirmware/Marlin #16307: [FR] [bugfix-2.0.x] Acquire position / don't move XY before parking nozzle

wikolii, 2019-12-23:
### Description

I don's consider this a bug, but there's definitely an issue with the filament change command "M600" combined with NOZZLE_PARK_FEATURE and ADVANCED_PAUSE_FEATURE when the motor drivers have been disabled and the axis moved manually (Position unknown). This will cause sometimes X and Y to hit their limits and bump creating noises and possibly breaking something.
I think it would be wise to add a feature to either home X and Y before parking the nozzle always or home X and Y only in case the XY position is unknown (motors disabled).

### Steps to reproduce

1. X and Y positions have to be greater than the NOZZLE_PARK_POINT
2. Issue an M84 command: Stop idle hold / Motors Off. Or even let it timeout according to FW settings. (This is most commonly the case after a print job is finished or cancelled, the motors are disabled and one moves the axis by hand)
3. Issue an M600 command: Filament change pause
4. The printer will try to go further than the axis limits bumping into endstops switches, etc.

### Feature Workflow

1. Add a checkpoint: if the position is unknown before parking, either home XY or don't move the axis at all, just raise Z. One can move the XY axis by hand if necessary
2. Enjoy the music of the axis not hitting their ends.

thisiskeithb, 2019-12-23:
You didn't include a zip file with your config files, but you need to enable `HOME_BEFORE_FILAMENT_CHANGE` in `Configuration_adv.h`.

thinkyhead, 2019-12-24:
While in principle it makes sense, it seems prudent that is should be optional, simply because endstops are not always the most positionally-accurate things, and at the same time leaving steppers powered on tends to heat them up more than keeping them moving, so both not-homing and not-staying-powered are often a preferred combination.

wikolii, 2019-12-24:
Thanks for the responses. I added the required files

> You didn't include a zip with your config files, but you need to enable `HOME_BEFORE_FILAMENT_CHANGE` in `Configuration_adv.h`.

It's weird how i missed this feature, thanks. I also added the files

> While in principle it makes sense, it seems prudent that is should be optional, simply because endstops are not always the most positionally-accurate things, and at the same time leaving steppers [...]

I personally like the last option, not-homing and not-staying-powered, but this would't work during a print job. So not homing XY if position is unknown still looks like the best option here, imho.

thisiskeithb, 2020-04-23:
@wikolii: After some more testing, I created issue #17663 and then PR #17681 to prevent the hotend/bed from crashing into the frame if `HOME_BEFORE_FILAMENT_CHANGE` is enabled.

closed 2020-06-30 by github-actions[bot]