MarlinFirmware/Marlin #16308: [BUG] (short description)

elmolinger, 2019-12-23:
I have installed a SKR1.3 with TMC2209 stepper drivers. The steppers make a lot of noise when printing. I tried to set the current without success. If Stealtchop is deactivated, no difference can be seen either.

It sounds like the printer is accelerating violently and stuttering.
In a circular motion, the steppers make a noise that I know from the old a4498 drivers, but I don't expect that from the TMC2209. Before that, a MKS 1.4 8-bit board with TMC 2208 drivers was installed, which was a lot quieter with the same settings.

I am using a current Marlin 2.0bugfix from today.
The TMC library used is 0.6.1

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/3995764/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/3995765/Configuration_adv.h.txt)
Video

https://www.youtube.com/watch?v=bcKDEgzNxB4

Any help would be appreciate

teemuatlut, 2019-12-23:
First make sure there are no mechanical issues.
After that you can try enabling Classic Jerk and disabling TMC monitoring and hybrid threshold.

Scope666, 2019-12-23:
On the recent bugfix I had periodic noise during faster travel moves, which when I heard it meant I had skipped steps.  The fix for me was rolling back to an older version that had Square Wave Stepping and Linear Advanced disabled.  These 2 are both enabled by default on my SKR Mini E3 1.2 w/ TMC2209's.  Just something to try to see if it makes a difference.

elmolinger, 2019-12-27:
Thank you for your prompt reply.

> 
> 
> First make sure there are no mechanical issues.
> After that you can try enabling Classic Jerk and disabling TMC monitoring and hybrid threshold.

I tried that, but to no avail. The noise stays the same.
I can also rule out mechanical errors. I recently renewed and checked the bearings. These run easily and quietly.

> On the recent bugfix I had periodic noise during faster travel moves, which when I heard it meant I had skipped steps. The fix for me was rolling back to an older version that had Square Wave [...]

Linear Advance was not activated for me. Which older version did you use?
Do newer versions work better for you now?

In the meantime I have increased the voltage on the Vref Trimpot of the TMC 2209 to 1.8Volt.
The stutter has now become quieter. I have installed 0.9Grad E3D stepper motors. I assume they need a little more voltage in UART mode, I just don't know how much.

A recent Video
https://youtu.be/mUXlkcg853k

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/4004761/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/4004762/Configuration_adv.h.txt)

mohacs, 2020-01-02:
Hey @elmolinger, 
I was having same problem, mostly from extruder motor. As soon as commented /#define STEALTHCHOP_E.  and Lin_advance, just fan noise nothing else.

boelle, 2020-01-07:
@elmolinger still an issue?

elmolinger, 2020-01-09:
The error persists.
Unfortunately I didn't have time to continue.

All changes in the firmware were unsuccessful and before I continue here, I decided to replace the stepper motors for the X and Y axes with standard 1.8 degree steppers.

I'll get in touch.

boelle, 2020-01-14:
@elmolinger i assume that the issue is still there and replacing steppers did not help?

elmolinger, 2020-01-19:
> 
> 
> @elmolinger i assume that the issue is still there and replacing steppers did not help?

Sorry, I don't have time for that right now. I open a new ticket as soon as I have exchanged the steppers and report and there are problems going on.

Thank you

closed 2020-01-19 by elmolinger