MarlinFirmware/Marlin #16310: [BUG] LPC176x P0.02 / P0.03 (UART0) not working when used as GPIO

Windaub57, 2019-12-23:
### Bug Description

LPC176x P0.02 / P0.03 (UART0) not working when used as GPIO

### My Configurations

stock configuration with any LPC176x boards
PIN P0.02 or P0.03 used as GPIO, i.e : enable pin 

### Steps to Reproduce

1. setup any board for LPC176x
2. setup use of PIN P0.02 or P0.03, i.e : axis enable

**Expected behavior:** pin state should change when enabling / disabling the motors
State were cross checked by using "M43 I" command.

**Actual behavior:** pins states remain high 

#### Additional Information

pins can however be controlled manually, using M42 command.

thinkyhead, 2019-12-24:
On AVR we typically can't use Pins 0 and 1 because they are hard-wired to the USB UART. I wonder if some boards based on the LPC1768 are the same way…?

ManuelMcLure, 2019-12-24:
I'm currently using ``P0_02`` on my Re-ARM to talk to my TMC2209s and that is using software serial, so I don't see any reason it wouldn't work for GPIO.

gloomyandy, 2019-12-24:
What exactly does "stock configuration with any LPC176x boards" mean? Please post your configuration files along with any changes you have made to the pins files or others (as requested in the bug report template). Also what version of Marlin are you using (and does it come from the official Marlin repository or are you using something from a manufacturers repository). What actual control board are you using?

Windaub57, 2019-12-24:
I used latest official Marlin Bugfix from 23/12/2019.
Stock configuration means I made no changes to the provided files, except from the MB (any LPC MB) (and PINS_DEBUGGING to have M43 available).
Then I assigned PIN 0.02/0.03 to X Enable (as example) and checked status change with M43, got no reaction.
Same when tested on custom development board using a LPC1769 uC, pin stay HIGH. Only usage of M42 can make it LOW.

gloomyandy, 2019-12-24:
So does that mean you still have:
```
#define SERIAL_PORT 0
```
In which case you are enabling the hardware serial port on pins P0.02 and P0.03. It really is easier if you actually post the configuration files as we can actually check for things like this without the 20 questions.

Windaub57, 2019-12-24:
Effectively, I also modified this one.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3998139/config.zip)

boelle, 2020-01-07:
@Windaub57 still having a problem?
@gloomyandy did the config posted say anything?

boelle, 2020-01-14:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-14 by boelle

Windaub57, 2020-01-14:
I changed on those two pins the usage of LPC176x::gpio_set(IO, V) to digitalWrite(IO, V) which is doing the following :
  LPC176x::gpio_set(pin, pin_status);
  // Set pin mode on every write (Arduino version does this)
  LPC176x::pin_enable_function(pin, LPC176x::Function::GPIO);
  LPC176x::gpio_set_output(pin);
Like this, I have no more issue.