MarlinFirmware/Marlin #16311: [BUG] LPC176x TMC SOFTWARE SPI too fast

Windaub57, 2019-12-23:
### Bug Description

TMC sorftware SPI is running too fast and 40bits setup messages not fully sent to drivers, resulting in random and anarchic driver configuration.
Also drivers reporting is fully corrupted.

### My Configurations

Stock configuration, LPC176x boards, TMC2130 SPI drivers

### Steps to Reproduce

1. Setup any LPC176x boards with TMC SOFT SPI DRIVERS

**Expected behavior:** proper communication with driver, setup and feedback

**Actual behavior:** complete random communication, drivers not properly setup, feedback corrupted

#### Additional Information

used a logic analyser to check messages sent. Modified writeSCK_H and writeSCK_L timing in TMC_platforms.h to make it slower and got proper communication back. See attached photos before after.

[image]
[image]

thinkyhead, 2019-12-24:
That file belongs to the `TMCStepper` repository, which is a different project under a different organization. Are you asking the maintainers of this project to head over there and fix `TMCStepper`? Or is there something that we can do here in our codebase to make `TMCStepper` behave as you are suggesting?

sjasonsmith, 2019-12-24:
This sounds like something @p3p fixed recently. Make sure you are using the latest version of TMCStepper.

Windaub57, 2019-12-24:
I do not have a full overview of Marlin to say if that could be a TMCStepper or Marlin issue. But if you think that 's only a TMCStepper, you can close it here and keep it for info.

For info, I used TMCStepper 0.6.1, latest Marlin (from yesterday) as well as latest LPC176x lib.

sjasonsmith, 2019-12-24:
> For info, I used TMCStepper 0.6.1, latest Marlin (from yesterday) as well as latest LPC176x lib.

The fixes I referred to are in 0.6.1, so it sounds like they must need adjusted a bit more.

It seems you've already figured out what needs to change. I'd suggest you close it here and post a pull request to the TMCStepper project, along with your traces that back up the need for the change.

Windaub57, 2019-12-24:
I'll do that!

closed 2019-12-24 by Windaub57