MarlinFirmware/Marlin #16312: [no FR, no BUG !] Advanced pause parameters clarification in Configuration_adv.h

yet-another-average-joe, 2019-12-24:
This is NOT a bug report nor a feature request. It's about the way the advanced pause parameters are organized in Configuration_adv.h : it seems to me that these parameters are not in a chronological/logical order, if I correctly understand the advanced pause. I didn't find much information about this feature, and it is mandatory with some extruders : BMGs are very difficult to load/unload by hand.

(it seems a bit diffrent from old #10873)

After reading pause.cpp, when a M600 command is issued :

Marlin does an initial retract of PAUSE_PARK_RETRACT_LENGTH @ PAUSE_PARK_RETRACT_FEEDRATE

then the hotend is parked

retract of FILAMENT_UNLOAD_RETRACT_LENGTH @ PAUSE_PARK_RETRACT_FEEDRATE mm/s
wait FILAMENT_UNLOAD_DELAY until the filament tip cools down and solidifies
ram purge the molten material pushing FILAMENT_UNLOAD_RETRACT_LENGTH (= unretract)  + FILAMENT_UNLOAD_PURGE_LENGTH @ maximum speed
then unload FILAMENT_CHANGE_UNLOAD_LENGTH of filament @ FILAMENT_CHANGE_UNLOAD_FEEDRATE and FILAMENT_CHANGE_UNLOAD_ACCEL

The UI asks for the user to change filament

the extruder gently grabs filament, pulling FILAMENT_CHANGE_SLOW_LOAD_LENGTH @ FILAMENT_CHANGE_SLOW_LOAD_FEEDRATE
then it pushes FILAMENT_CHANGE_FAST_LOAD_LENGTH @ FILAMENT_CHANGE_FAST_LOAD_FEEDRATE and FILAMENT_CHANGE_FAST_LOAD_ACCEL

purge:
purge ADVANCED_PAUSE_PURGE_LENGTH @ ADVANCED_PAUSE_PURGE_FEEDRATE
goto purge: again if needed

Understanding this procedure is important for the user to optimize the parameters by hand with some gcode before compiling/flashing.

If I understand correctly the code and #defines, maybe it would be easier to understand and customize Advanced Pause Parameters if the code looked like that, with very few more comments :

**(Github completely messed up source formatting, see attached file...)**

`#define ADVANCED_PAUSE_FEATURE
#if ENABLED(ADVANCED_PAUSE_FEATURE)
 // This short retract is done immediately, before parking the nozzle.
  #define PAUSE_PARK_RETRACT_LENGTH            2  // (mm) Initial retract.
  #define PAUSE_PARK_RETRACT_FEEDRATE         60  // (mm/s) Initial retract feedrate.

 // nozzle is parked : Filament Unload does a Retract, Delay, and Purge first:
  #define FILAMENT_UNLOAD_RETRACT_LENGTH      13  // (mm) Unload initial retract length.
  #define FILAMENT_UNLOAD_DELAY             5000  // (ms) Delay for the filament to cool after retract.

// ram purge
  #define FILAMENT_UNLOAD_PURGE_LENGTH         8  // (mm) An unretract is done, then this length is purged.

// unload
  #define FILAMENT_CHANGE_UNLOAD_LENGTH      100  // (mm) The length of filament for a complete unload.
												//   For Bowden, the full length of the tube and nozzle.
												//   For direct drive, the full length of the nozzle.
												//   Set to 0 for manual unloading.
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE     10  // (mm/s) Unload filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_UNLOAD_ACCEL        25  // (mm/s^2) Lower acceleration may allow a faster feedrate.

// reload
  #define FILAMENT_CHANGE_SLOW_LOAD_LENGTH     0  // (mm) Slow length, to allow time to insert material.
												// 0 to disable start loading and skip to fast load only
  #define FILAMENT_CHANGE_SLOW_LOAD_FEEDRATE   6  // (mm/s) Slow move when starting load.
  #define FILAMENT_CHANGE_FAST_LOAD_LENGTH     0  // (mm) Load length of filament, from extruder gear to nozzle.
  #define FILAMENT_CHANGE_FAST_LOAD_FEEDRATE   6  // (mm/s) Load filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_FAST_LOAD_ACCEL     25  // (mm/s^2) Lower acceleration may allow a faster feedrate.
												  //   For Bowden, the full length of the tube and nozzle.
												  //   For direct drive, the full length of the nozzle.

// purge before resume printing
  //#define ADVANCED_PAUSE_CONTINUOUS_PURGE       // Purge continuously up to the purge length until interrupted.
  #define ADVANCED_PAUSE_PURGE_FEEDRATE        3  // (mm/s) Extrude feedrate (after loading). Should be slower than load feedrate.
  #define ADVANCED_PAUSE_PURGE_LENGTH         50  // (mm) Length to extrude after loading.
												  //   Set to 0 for manual extrusion.
												  //   Filament can be extruded repeatedly from the Filament Change menu
												  //   until extrusion is consistent, and to purge old filament.

#define ADVANCED_PAUSE_RESUME_PRIME          0  // (mm) Extra distance to prime nozzle after returning from park.
//#define ADVANCED_PAUSE_FANS_PAUSE             // Turn off print-cooling fans while the machine is paused.

#define PAUSE_PARK_NOZZLE_TIMEOUT           45  // (seconds) Time limit before the nozzle is turned off for safety.
#define FILAMENT_CHANGE_ALERT_BEEPS         10  // Number of alert beeps to play when a response is needed.
#define PAUSE_PARK_NO_STEPPER_TIMEOUT           // Enable for XYZ steppers to stay powered on during filament change.

//#define PARK_HEAD_ON_PAUSE                    // Park the nozzle during pause and filament change.
//#define HOME_BEFORE_FILAMENT_CHANGE           // Ensure homing has been completed prior to parking for filament change

//#define FILAMENT_LOAD_UNLOAD_GCODES           // Add M701/M702 Load/Unload G-codes, plus Load/Unload in the LCD Prepare menu.
//#define FILAMENT_UNLOAD_ALL_EXTRUDERS         // Allow M702 to unload all extruders above a minimum target temp (as set by M302)
#endif
`

boelle, 2019-12-24:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-12-24 by boelle

yet-another-average-joe, 2019-12-24:
It was not a question or help request !

It was a suggestion for a reorganization of Configuration_adv.h, with edited code for the section  "Advanced Pause" in the attachment : it was definitely development related, not support related. And I was not asking for something, I was describing the reasons why it should be reorganized.

boelle, 2019-12-24:
we only have bugs and fr's on the issue list, rest needs to go elsewhere

Vertabreak, 2019-12-24:
to make code change suggestions file a pull request.

boelle, 2019-12-24:
or go to discord and mention your idea

boelle, 2019-12-24:
loads of options that do not include the issue list

yet-another-average-joe, 2019-12-24:
Unfortunately, I have a huge problem with Windows GitGUI and I can't even do commits on my own Git repositories ! (maybe the paranoid settings I put in my FW) Never solved the problem ! A few months ago I slightly edited Marlin code, I posted the code here, and someone did a pull request after he added enhancements. 

If RepRap forum is the place to go, and someone can do pull requests, I will subscribe.

aleung, 2020-05-08:
@yet-another-average-joe thanks. Finally understood the procedure and how to configure.