MarlinFirmware/Marlin #16313: First print line for cleaning nozzle began in oppsite side of bed

jack38497, 2019-12-24:
### Bug Description

Printing the first two lines for cleaning the nozzle starting in the opposite side of the bed. Before upgrade it began at the right corner now it begins at the lower left corner. 

### My Configurations

3d printer: Creality Ender-5
Slicer App: Cura 4.4.1
Config ZIP file: [Marlin20x_Ender5.zip](https://github.com/MarlinFirmware/Marlin/files/3996848/Marlin20x_Ender5.zip)

### Steps to Reproduce

1. Upgrade Ender-5 with Marlin 2.0.x.
2. Slice stl test cube with Cura 4.4.1.
[xyzCalibration_cube.zip](https://github.com/MarlinFirmware/Marlin/files/3996866/xyzCalibration_cube.zip)
3. Save gcode in microSD.
4. Print. 

**Expected behavior:** [What you expect to happen]

Print the first two lines starting at the upper right side of the bed.
Ends print and position nozzle in home corner(upper right).

**Actual behavior:** [What actually happens]

Print the first two lines starting at the lower left side of the bed.
Ends print and position nozzle in opposite home corner(lower left).

#### Additional Information

[image]

thisiskeithb, 2019-12-24:
Creality ships the Ender-5 with incorrect settings like homing to X and Y min which causes everything to be mirrored.

In reality, the Ender-5 homes to X and Y max, so it’s been corrected in the Marlin 2.0 config.

jack38497, 2019-12-24:
Before the upgrade there was no issue. As I saw the piece in Cura 4.4.1 it was printed in that position. I even printed a four pieces sword before the upgrade. 

In the image the piece is at the correct position but the first two lines are in the opposite side of the bed. The cube has the letters XYZ.

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

thisiskeithb, 2019-12-24:
> btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

This will be the same for anyone upgrading since everything is mirrored from factory due to incorrect  `USE_*_PLUG` & `*_HOME_DIR` settings.

boelle, 2019-12-24:
is there anything we can do about it other than changing the example configs?

sjasonsmith, 2019-12-24:
This is a change from the old factory configuration, but it is a good change. It makes the printer treat front left as (0,0), like virtually every other printer out there. The factory method was to act like the printer was sitting on your desk facing away from you, which was very confusing.

In my opinion, there is no issue here. Just a behavior change to get used to, and appreciate if you are used to working with other printers already.

thisiskeithb, 2019-12-24:
I agree that changing Marlin’s configs to match incorrect factory settings is not a good idea.

boelle, 2020-01-08:
@thisiskeithb @sjasonsmith so would it make most sense here to close and say we will not fix as the factory settings are wrong?

sjasonsmith, 2020-01-08:
@boelle, I think it should be closed.

The BTT config makes the (0,0) point be in the rear right corner of the printer. This is their choice in their fork.

In mainline Marlin, I think all printers should consistently make (0,0) be the front left, as the examples do now. Using the Ender 5 with the BTT layout was super frustrating for me, and I would have made this change myself had it not already been done in Marlin.

closed 2020-01-10 by boelle