MarlinFirmware/Marlin #16314: [BUG] Screen doesn't turn on on skr 1.3 with cr10-S screen

Liger0, 2019-12-24:
### Bug Description

The screen doesn't turn on on a skr 1.3 with the doble ribbon cable cr10-S screen

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3996943/Marlin.zip)

### Steps to Reproduce

1. Flash the firmware
2. Start the printer

**Expected behavior:** 
LCD should work

**Actual behavior:** 
LCD doesn't work

#### Additional Information

As what I heard the pins are reversed on the skr 1.3 exp1 and exp2k, but I guess Marlin could account for it?

ellensp, 2019-12-24:
The pins are flipped 180 degrees for some displays...  and no you can't correct for it in firmware as the power pins can't be changed.  You need a custom cable or to remove the keys on the cable so you can plug them in rotated 180, or cut new key slots. Or remove the plastic with the key slots from the back of the display.

Liger0, 2019-12-24:
Could there be a comment in the FW at least at the LCD line? Since I just barely read about it by pure luck and it isn't written anywhere. I really struggled to find anything without any luck.

Liger0, 2019-12-24:
I did cut the pin on one side of each wire and now it turns on. 
The issue is that, before the marlin logo, this happens:
[image]

ellensp, 2019-12-24:
Yes that is a known issue, give it  30 seconds and marlin will finish booting.

ellensp, 2019-12-24:
see issue #15855

boelle, 2019-12-24:
@ellensp so both this isse and #15855 could be closed?

Liger0, 2019-12-24:
Could you ass a disclaimer about invert the pins please? People aren't omniscient.

ellensp, 2019-12-24:
@boelle This issue definitely. His screen is working, and it wasn't a firmware issue. As for 15855 After replacing he bootloader the issue is still gone for me...

closed 2020-01-07 by boelle