MarlinFirmware/Marlin #16316: PID for Chamber

Nazrac, 2019-12-24:
I think PID for chamber should be added. I have looked into the code and think I could get it to work, but would rather let someone more experienced than myself do it. If I am right it would only take 5 minutes, but  if I am wrong I would mess the whole think up.

thisiskeithb, 2019-12-24:
Give it a try! If it works, submit a PR and everyone can review it.

RaabenFuchs, 2020-06-28:
thanks. exaactly what i was looking for. I just startet copying the PID from the bed.

i'm heating my enclosure with the heatsink like this issue hints https://github.com/MarlinFirmware/Marlin/issues/18295
and going to post it on thingiverse as soon as this PID thing works.

compiling the feature was ok, but not sure on how to do the store and load from eeprom part. and do the menue items.. what else is required for integration? lets see, if it works out. 

@Nazrac or are you heading any further, let me know

ksanislo, 2021-02-21:
So, I decided to take a stab at this myself this afternoon and I've got a working implementation, but it's still pretty rough around the edges, and based off 2.0.7.2 directly and not bug fix.

I also had to "invent" a new M309 command for chamber tuning, similar to M304 for bed. I'm not 100% sure this is the best approach, but as I've done literally zero marlin work previously, I'm fully open to suggestions if someone has a better approach. I also skipped adding the menus for the LCD, but putting them in shouldn't be too much more work.

You can auto-tune with the expected command along the lines of "M303 E-2 C5 S42" to get your tuning values.

You can see my example just in https://github.com/ksanislo/Marlin directly, but I can put together a proper patch if there's any interest.

RaabenF, 2021-02-21:
Nice,
i'm working on that, too:
https://github.com/RaabenF/Marlin/commits?author=RaabenF

thank you, i did it really the same way but copied some minor stuff from you. You can have a look at my menues if you like.
https://github.com/RaabenF/Marlin/commit/d2461f2ead5249240372cd98bdd7edbc8102f41c

I have issues with my switches interrupts, and till now the storage hasnt worked
this is my usecase:
https://www.thingiverse.com/thing:4770836

so you can imagine, why i try to do a "one fan" approach
i didnt try on autotune i think this is not necessary on a chamber, and i dont youse gcode either.

i am a bit scared about doing all the language translation

ksanislo, 2021-02-21:
@RaabenF I'm not completely sure I understand what you're trying to accomplish with the one fan setup, but my patch should provide a solid starting point with functional PID. Autotune is definitely worth using, with my chamber PID tuned, I have ~+/- 0.5c variance in temps, but untuned I was getting substantial swings in temperature (approx. +/- 5c) while printing.

I did already look at the menus, and will probably go add them in the next few days, but the existing design has plenty of poor assumptions made about the heater setup and didn't seem particularly flexible. Fixing it properly will require adding in a bunch of additional settings like being able to define preset chamber temps and the like first, but all that seemed a bit more optional to me at the time as I typical run my printer via octoprint and just use the terminal with gcode commands as required.

 I'm currently back to printing my new bgdog modular case (The PID temp control was really because I needed to fix a warping problem on large panels.), so I'm out of ability to test further for another ~24 hours for my print to finish before I can resume development on the patch.

RaabenF, 2021-02-21:
#response_container_BBPPID{font-family: initial;font-size: initial;color: initial;}  Yes, having presets would be great, though not necessary.The more i got into marlin, the less i liked it, and when my endstops stopped working, i somehow refused.I assumed +-5C would be ok, but heating to 80C simply isnt possible with a static heater. YesFor my purpose i inverted the output - the slower my fan is spinning, the hotter my air from the heater gets - wich gives me always like 10 degree UNDER target temp. But ok. I thought it is easier to use one blower fan: it cools the heatsink then the same airstream can be used to cool the print. No second part-cooling fan required. (For PLA)OR: at very slow fan speeds you get hot air out of the heatsink to heat that chamber. (For PC)Sry, dont know what a 'bgdog' is!? Have fun with the menus! Im curious how long u take, that crazy macro code is just insane!                                                                                                                                                                                                                                                                                                                Von: notifications@github.comGesendet: 21. Februar 2021 20:36An: Marlin@noreply.github.comAntworten: reply@reply.github.comCc: dhilipp@web.de; mention@noreply.github.comBetreff: Re: [MarlinFirmware/Marlin] [FR] PID for Chamber (#16316)  
@RaabenF I'm not completely sure I understand what you're trying to accomplish with the one fan setup, but my patch should provide a solid starting point with functional PID. Autotune is definitely worth using, with my chamber PID tuned, I have ~+/- 0.5c variance in temps, but untuned I was getting substantial swings in temperature (approx. +/- 5c) while printing.
I did already look at the menus, and will probably go add them in the next few days, but the existing design has plenty of poor assumptions made about the heater setup and didn't seem particularly flexible. Fixing it properly will require adding in a bunch of additional settings like being able to define preset chamber temps and the like first, but all that seemed a bit more optional to me at the time as I typical run my printer via octoprint and just use the terminal with gcode commands as required.
I'm currently back to printing my new bgdog modular case (The PID temp control was really because I needed to fix a warping problem on large panels.), so I'm out of ability to test further for another ~24 hours for my print to finish before I can resume development on the patch.

ksanislo, 2021-02-21:
Oh, get what you're doing now, that's a pretty neat idea.

[This is the "bgdog" case](https://github.com/bgdog1/3dPrinterCase) I've been printing that was having warping problems previously. It's been running good all day now though.

I'm not really sure how much time I put into it, maybe ~4 hours. It certainly wasn't the 5 minutes suggested in the original comment. I tried to keep things as similar to the original code in intent as possible, hopefully it doesn't break anybody's edge-case use of the previous cooling functions.

RaabenF, 2021-02-22:
" wasn't the 5 minutes" i'm sure the guy noticed
"I tried to keep things as similar" - you seem more experienced with projects. Is this how it always goes?
i've spent some hours in the discord, but didnt really seem helpfull too. the guys there hanging around.

Please tell me if your interrupts and the autotune and all really works.
i rember the "autotune" menu button selects the heater automatically, not sure, are you planning to do a second button?

as you can see, i did the thing on the bugfix branch, which means you could do a pull with that. "temperature.cpp" had at least one change since, that affects our matter directly. And all macros have changed from stable to bugfix (TERN()...is used for pre processor defines i think)
if that matters to you, you could use my code, since it's basically the same as yours. because you seem to get things quite faster then me and i really dont know, how to take it further to be suitable for anyone. i needed days for all this to figure out.

thisiskeithb, 2021-02-25:
Merged.

closed 2021-02-25 by thisiskeithb