MarlinFirmware/Marlin #16322: [BUG] MKS Gen L pins not remapping

LordBBQ, 2019-12-25:
I have got my hands on a MKS Gen L board. There is an issue with the PCB in that as soon as there is any load on it, the MOSFET blows. This happened on 2 boards! I made up an external MOSFET. I could not remap the pins to the servo header as it does not work. How do I do this?

ellensp, 2019-12-25:
On two boards.., your most likely trying to drive a heated bed that is way larger than the MOSFET can handle.  And you just edit the pins.h file your using which is dependent on what Motherboard you have defined. This should be BOARD_MKS_GEN_L which leads to pins_MKS_GEN_L.h  but this file includes pins_RAMPS.h so you need to edit this file. Find the line #define RAMPS_D8_PIN      8
 and change it to #define RAMPS_D8_PIN  11.  you also need to change #define SERVO0_PIN     11 to #define SERVO0_PIN     -1    I suspect this is what you did and It ddn't work as your using as 12v external mosfet on a 5v logic pin.

LordBBQ, 2019-12-25:
Thanks, did not remap the servo1 pin. The mosfet uses low side switching, and worked fine when manually connecting the gstepin to 5v or ground.

LordBBQ, 2019-12-25:
The pin is being driven now, but the MOSFET is not actually putting enough power through the bed. At one stage, it heated fine to 38C, but then could not heat anymore, and dropped down and triggered thermal runaway. This MOSFET says it has an input voltage of ±20v. Am I correct in saying it can be driven by the board's logic?

sjasonsmith, 2019-12-28:
Is there still any concern there is a Marlin issue here? It looks like the remaining question is just regarding an external MOSFET's current draw, and nothing Marlin-specific.

ellensp, 2019-12-28:
nothing marlin specific, and no info provided on the mosfet...

LordBBQ, 2019-12-28:
The board is now driving the mosfet, still not working properly. I am returning the faulty boards. No longer a marlin issue. Thanks for the help anyway.

closed 2019-12-28 by LordBBQ