MarlinFirmware/Marlin #16323: X and Y axes move in wrong direction after homing. (marlin 2.0 on skr 1.3 with tmc 2130)

XOIIO, 2019-12-25:
### Bug Description

Hi all, so this is probably an issue with my configuration but maybe also a bug, I am upgrading my printer to an skr 1.3 and I haven't used TMC2130's before, so I have a feeling this issue relates to my use of sensorless homing.

Essentially, all my axes move properly, and auto homing works properly, however, if you try to perform an auto bed level, or home the printer a second time, the x and y axes both go the wrong direction, colliding with the frame, and the tmc2130 does not trigger a stop.

I've tried fresh builds and every time I load the firmware on this behavior is present. I'm guessing there's a simple configuration mistake that I'm making however I can't spot what it would be.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4000244/Marlin.zip)

### Steps to Reproduce

1. Home printer
2. home printer or perform automatic bed leveling

**Expected behavior:** [What you expect to happen]

printer should either home again properly, or move to the front left of the bed.

**Actual behavior:** [What actually happens]

printer seems to be running x and y axes in reverse when performing an automatic bed level, or auto home a second time.

https://youtu.be/5-gNjRE9ot0

boelle, 2020-01-07:
@XOIIO still an issue?

XOIIO, 2020-01-08:
@boelle Yes, I never found a fix for this, however once I went back to microswitches for end stops, this behavior stopped as well. It seems that for some reason enabling sensorless homing inverts where the bed mapping begings, and I guess the overall bed position, or something along those lines.

boelle, 2020-01-10:
very strange, and it does not matter if you use 2.0.1 or bugfix 2.0.x ?

XOIIO, 2020-01-12:
> very strange, and it does not matter if you use 2.0.1 or bugfix 2.0.x ?

I'm on the bugfix release, I'll try to remember to change version and test it out later today. Will be at least 12hrs from this comment.

boelle, 2020-01-18:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-18 by boelle