MarlinFirmware/Marlin #16324: Bigtreetech SKR 1.4pro

Docomar750, 2019-12-25:
Hi

Trying to compile for the SKR1.4PRO, this uses a LPC1769 processor whats the best why to to copy the pins.h from LPC1768 to LPC1769 and link it to the BIGTREETECH SKR 1.4 board or do i have to define it as BIGTREETECH SKR1.4PRO ?

ellensp, 2019-12-25:
Seem to be identical to SKR 1.4 but using the faster 120mhz lpc1769 
Need to create a new board..  Just compiling currently... if it works ill share a link shorty for you to try, since I don't have this board.

ellensp, 2019-12-25:
grab marlin bugfix-2.0.x from https://github.com/ellensp/Marlin In platformio.ini set default_envs = LPC1769 In Configuration.h set #define MOTHERBOARD BOARD_BIGTREE_SKR_V1_4_PRO  and #define SERIAL_PORT -1 and give this a try. If it works, ill turn this into a PR for all..

ellensp, 2019-12-25:
Board also seems to be called BTT SKR V1.4 turbo in various places.

thisiskeithb, 2019-12-25:
> Board also seems to be called BTT SKR V1.4 turbo in various places.

It's been officially renamed to the SKR 1.4 Turbo to avoid confusion with the SKR 1.1 Pro.

ellensp, 2019-12-26:
Turbo does make more sense...  updated and cleaned up pins files. #define MOTHERBOARD now needs BOARD_BIGTREE_SKR_V1_4_TURBO

Docomar750, 2019-12-26:
Hi All
Thanks for all the assistance it's a great help got my head around mechanical aspect of 3d print now trying to understand the coding better.

Yes I can see the name change as TURBO is what's advertised on Bigtreetech site now.

Darren

captainfresh, 2019-12-27:
Looks like it has issues elsewhere??
I managed to get it to compile by just changing pins_BTT_SKR.h to the following..
`#ifndef MCU_LPC1768
  #ifndef MCU_LPC1769
    #error "Oops! Make sure you have the LPC1768 or LPC1769 environment selected in your IDE."
  #endif
#endif`

Its very likely that its a dumb approach as im a complete noob at this stuff but it seems to work...

The errors when using BOARD_BIGTREE_SKR_V1_4_TURBO are shown below...

`Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip; board: nxp_lpc1769; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x 0.1.1 > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1), framework-arduino-lpc176x 0.2.2
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0        
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4.1   
|-- <TMCStepper> 0.6.1   
|-- <Adafruit NeoPixel> 1.3.0
|-- <SailfishLCD>
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1769/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o
In file included from c:\skrturbo\marlin\src\inc\MarlinConfigPre.h:37,
                 from c:\skrturbo\marlin\src\inc\marlinconfig.h:28,
                 from c:\skrturbo\marlin\src\core\serial.h:24,
                 from Marlin\src\HAL\HAL_LPC1768\DebugMonitor.cpp:26:
c:\skrturbo\marlin\configuration.h:133: warning: "BOARD_BIGTREE_SKR_V1_4_TURBO" redefined
   #define BOARD_BIGTREE_SKR_V1_4_TURBO  // BigTreeTech SKR v1.4 PRO (Power outputs: Hotend0, Hotend1, Fan, Bed)

In file included from c:\skrturbo\marlin\src\inc\MarlinConfigPre.h:35,
                 from c:\skrturbo\marlin\src\inc\marlinconfig.h:28,
                 from c:\skrturbo\marlin\src\core\serial.h:24,
                 from Marlin\src\HAL\HAL_LPC1768\DebugMonitor.cpp:26:
c:\skrturbo\marlin\src\core\boards.h:225: note: this is the location of the previous definition
 #define BOARD_BIGTREE_SKR_V1_4_TURBO    2500  // BigTreeTech SKR v1.4 PRO (Power outputs: Hotend0, Hotend1, Fan, Bed)

In file included from c:\skrturbo\marlin\src\inc\MarlinConfigPre.h:37,
                 from c:\skrturbo\marlin\src\inc\marlinconfig.h:28,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
c:\skrturbo\marlin\configuration.h:133: warning: "BOARD_BIGTREE_SKR_V1_4_TURBO" redefined
   #define BOARD_BIGTREE_SKR_V1_4_TURBO  // BigTreeTech SKR v1.4 PRO (Power outputs: Hotend0, Hotend1, Fan, Bed)

In file included from c:\skrturbo\marlin\src\inc\MarlinConfigPre.h:35,
                 from c:\skrturbo\marlin\src\inc\marlinconfig.h:28,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
c:\skrturbo\marlin\src\core\boards.h:225: note: this is the location of the previous definition
 #define BOARD_BIGTREE_SKR_V1_4_TURBO    2500  // BigTreeTech SKR v1.4 PRO (Power outputs: Hotend0, Hotend1, Fan, Bed)

Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\main.cpp.o
In file included from c:\skrturbo\marlin\src\inc\MarlinConfigPre.h:37,
                 from c:\skrturbo\marlin\src\inc\marlinconfig.h:28,
[... 256 lines cut ...]
c:\skrturbo\marlin\src\inc\SanityCheck.h:1498:4: error: #error "HEATER_0_PIN not defined for this board."
   #error "HEATER_0_PIN not defined for this board."
    ^~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\LPC1769\src\src\HAL\HAL_LPC1768\main.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\HAL_LPC1768\timers.cpp.o] Error 1
======================================================================================== [FAILED] Took 2.63 seconds ========================================================================================

Environment                   Status    Duration
----------------------------  --------  ------------
megaatmega2560                IGNORED
megaatmega1280                IGNORED
rambo                         IGNORED
FYSETC_F6_13                  IGNORED
sanguino_atmega644p           IGNORED
sanguino_atmega1284p          IGNORED
melzi                         IGNORED
melzi_optiboot                IGNORED
at90usb1286_cdc               IGNORED
at90usb1286_dfu               IGNORED
DUE                           IGNORED
DUE_USB                       IGNORED
DUE_debug                     IGNORED
LPC1768                       IGNORED
LPC1769                       FAILED    00:00:02.627`

ellensp, 2019-12-27:
@captainfresh what is in your configuration.h line 133  
Look like you have set #define BOARD_BIGTREE_SKR_V1_4_TURBO
not #define MOTHERBOARD BOARD_BIGTREE_SKR_V1_4_TURBO

captainfresh, 2019-12-27:
Ah yes, you're quite right, apologies!

captainfresh, 2020-01-01:
I have managed to compile and get most things *working on a Tronxy X5SA Pro (core XY) but I did have to change Line 64 in pins_BTT_SKR.h as it seems to overwrite the pins_BTT_SKR_V1_4_TURBO.h for the heatbed temp sensor to work.
Changed:
```
#ifndef TEMP_1_PIN
  #define TEMP_1_PIN       P0_25_A2   // A2 (T2) - (69) - TEMP_1_PIN
#endif
#ifndef TEMP_BED_PIN
  #define TEMP_BED_PIN     P0_23_A0   // A0 (T0) - (67) - TEMP_BED_PIN
#endif
```
to

```
#ifndef TEMP_1_PIN
  #define TEMP_1_PIN       P0_23_A2   // A2 (T2) - (69) - TEMP_1_PIN
#endif
#ifndef TEMP_BED_PIN
  #define TEMP_BED_PIN     P0_25_A0   // A0 (T0) - (67) - TEMP_BED_PIN
#endif
```

The only thing to tackle now is when homing X or Y the Z moves up and the motors Z make a grinding noise at the same time. If I move the Z up and down on its own its ok. I have turned off SAFE_HOMING but its still the same. Both Z motors turn the same direction as expected. I also get an error about ABL being disabled on boot - not sure if its related...

Also as a heads up for fellow Tronxy owners. So far only the fan JST wiring needs to be reversed.

ellensp, 2020-01-01:
pins_BTT_SKR_V1_4_TURBO.h just sets a few defines and includes pins_BTT_SKR_V1_4.h which includes pins_BTT_SKR.h. It doesn't set any pins. Also   P0_23_A2 and  P0_25_A0 is not valid pins it needs to be P0_25_A2 and be P0_23_A0  as defined https://github.com/p3p/pio-framework-arduino-lpc176x/blob/master/system/lpc176x/pinmapping.h

captainfresh, 2020-01-02:
Yes sorry I meant that pins_BTT_SKR.h as it seems to overwrite the pins_BTT_SKR_V1_4.h not TURBO.
If I understand correctly, by making an invalid define within pins_BTT_SKR.h the defines in pins_SKR_V1_4.h were used? Either the way the point I was trying to make is that TEMP_BED_PIN is set to in pins_BTT_SKR_V1_4.h 
```#define TEMP_BED_PIN       P0_25_A2```

but in pins_BTT_SKR.h it is
 ```
#ifndef TEMP_BED_PIN
  #define TEMP_BED_PIN     P0_23_A0   // A0 (T0) - (67) - TEMP_BED_PIN
#endif
```

randellhodges, 2020-01-02:
`#ifndef` = if not defined, because it is defined in `pins_BTT_SKR_V1_4h` on line 111 and `pins_BTT_SKR.h` is included on line 116, that line you referenced in `pins_BTT_SKR.h` doesn't do anything.

Chill00r, 2020-01-05:
Hi, I updated the files like in #16374 

but i still have a prob to compile 

`Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\u8g\LCD_pin_routines.c.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/../pins/lpc1768/pins_BTT_SKR_V1_4.h:118,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/../pins/pins.h:369,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\HAL_LPC1768\DebugMonitor.cpp:26:
Marlin\src\HAL\HAL_LPC1768\../../core/../inc/../pins/lpc1768/pins_BTT_SKR.h:29:4: error: #error "Oops! Make sure you have the LPC1768 environment selected in your IDE."
   #error "Oops! Make sure you have the LPC1768 environment selected in your IDE."
    ^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/lpc1768/pins_BTT_SKR_V1_4.h:118,
                 from Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:369,
                 from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/lpc1768/pins_BTT_SKR.h:29:4: error: #error "Oops! Make sure you have the LPC1768 environment selected in your IDE."`

Chill00r, 2020-01-05:
Hello, I still have a problem to compile (not full text)

> Compiling .pio\build\LPC1769\src\src\HAL\HAL_LPC1768\u8g\LCD_pin_routines.c.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/../pins/lpc1768/pins_BTT_SKR_V1_4.h:118,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/../pins/pins.h:369,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\HAL_LPC1768\DebugMonitor.cpp:26:
Marlin\src\HAL\HAL_LPC1768\../../core/../inc/../pins/lpc1768/pins_BTT_SKR.h:29:4: error: #error "Oops! Make sure you have the LPC1768 environment selected in your IDE."
   #error "Oops! Make sure you have the LPC1768 environment selected in your IDE."
    ^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/lpc1768/pins_BTT_SKR_V1_4.h:118,
                 from Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:369,
                 from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/lpc1768/pins_BTT_SKR.h:29:4: error: #error "Oops! Make sure you have the LPC1768 environment selected in your IDE."

ellensp, 2020-01-05:
Please learn how to use pastebin . It looks like you have set motherboard to BOARD_BIGTREE_SKR_V1_4 not BOARD_BIGTREE_SKR_V1_4_TURBO

boelle, 2020-01-05:
closing as the board has been added

boelle, 2020-01-05:
locked as users with problems should create an issue and not add to a feature request 

PLEASE read the pinned post at top of issue list, and ReadTheFineManual (issue template)

closed 2020-01-05 by boelle