MarlinFirmware/Marlin #16327: No second bump after homing on Y axis.

GurenWolf, 2019-12-25:
Hello everyone.

I have a Anycubic Kossel Linear Plus, Biggtreetech SKR 1.3 board, TMC2208 drivers in UART mode and the latest marlin firmware with a fairly standard setup.

To mkae it short, I have had some issues with bed leveling. After close inspection, I noticed that after homing, X and Z axes do the slower second bump and then retract a little. This is not the case with the Y axis; it just does the first bump and doesn't retract after doing so. 

(Printer orientation: looking at it from the front, where the LCD panel is located, X axis is on the left, Y axis on the right, Z axis on the back.)

Yes, I have checked the endstops with a multimeter, I have checked the connections on the board and everything is alright, no broken wires, no faulty switches.

As usual, I start blaming it now on the firmware. I was trying to find the piece of code where the second bump and retraction value after the bump are located, but I had no luck., (and no, I don't mean #define Y_HOME_BUMP_MM 5, which is the distance traveled before the second bump, I mean the distance traveled after the second bump is done).

Any ideas on the issue or where I could find this piece of code inside the massive amount of files within the firmware?

Thanks beforehand.

boelle, 2019-12-25:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

GurenWolf, 2019-12-26:
You are right, I could've sworn I had uploaded them, sorry about that.

These are the files"
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4003308/Marlin.zip)

sjasonsmith, 2019-12-28:
I think you are seeing adjustments due to tower endstep height offsets, which can be set using M666.
http://marlinfw.org/docs/gcode/M666.html

I am guessing that if you use `M503 S0` to examine the current state of M666, Y will have zero offset, and the other two towers will have negative offsets.

I saw similar behavior with my printer, but if I clear those offsets with `M666 X0 Y0 Z0` then all towers stay on the switch after the second bump.

GurenWolf, 2019-12-28:
@sjasonsmith, you are right about that, if endstop offsets are set to 0, all of the rails keep the switches triggered. Now that got me wondering though, if these offset values are calculated after autocalibration, why are they so off?

I will test another print after changing these values to see if there is any improvement on the first layer, if that's not the case, I might open another thread regarding a leveling issue that I have been dealing with for quite a while, even after readjusting the printer mechanically and inside the firmware.

GurenWolf, 2019-12-28:
So, I did testing after changing these offset settings, didn't see any improvement on the first layer, it is way too low on one side and way to high on the other. It seems like the problem is something else.

I will be posting another thread regarding the issue with more in-depth. Thanks a lot to everyone.

closed 2019-12-29 by GurenWolf