MarlinFirmware/Marlin #16328: [BUG] (SD-card problem)

koreandi, 2019-12-25:
Hello dear community.
I need help to get my SD card running...
Please.. I am very new to Marlin and did get all the rest to work.
I have read many threads and tried many things...

### Bug Description

I am struggling since a few day with my printer to get my SD-Card work correctly.

### My Configurations

Creality CR10
SKR 1.3
TMC2208
E3d Hemera
Simplify3D 4.1.2
Marlin 2.0 (also Bugfixed version)
SD-Card Cable

### Steps to Reproduce

With USB connected the SD-Card is recognized on WIN10.
Copying files is possible.
Print with USB works fine.

Additional: When using SD-Card-Cable, the SD Card is not recognized on WIN10. --> I can live with the fact that the Micro-SD-Card is fix in the printer...

Micro-SD-Card is not recognized by Marlin/Printer.
Printer says: Media released; Init fails with message on S3d:
- READ: echo:SD init fail
- SD init fail

**Expected behavior:**
Print via USB should work

**Actual behavior:**
SD-Card 

#### Additional Information

SD-Card Cable: https://cutt.ly/brrmMBT

My Config files: https://drive.google.com/file/d/1-NjGYGE9a6aWHiYuSEaKHLay2s8_FpUS/view?usp=sharing

boelle, 2019-12-25:
try here first

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

shitcreek, 2019-12-26:
Please be sure that the SD card cable and SD card are properly seated. Make sure there isn't any tear in the SD card cable and the SD card works and is properly formatted.

boelle, 2020-01-07:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-07 by boelle