MarlinFirmware/Marlin #16329: [BUG] hard knock/shock crossing hybrid_threshold

rado79, 2019-12-25:
### Bug Description

hard knocking on travelmoves - each one time while accelerating/decelerating across hybrid_threshold.
travelmove speed is 200mm/s, acceleration 1200, hybrid_thershold testet 60-150mm/s.
the 1.8 stepper motors can handle this hard noisy moves, with 0.9 motors results all in layer shifts. tried with a set of TMC2208 and 2209.

### My Configurations

CoreXY
SKR 1.3
TMC2208 and TMX2209 activ cooling
24V
Marlin 2.0.1 and a lot of older bugfixes
Octoprint

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4000858/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4000859/Configuration_adv.zip)

### Steps to Reproduce

compile with #define HYBRID_THRESHOLD
start a testprint - travelmoves faster than hybrid threshold

**Expected behavior:** smooth travelmoves

**Actual behavior:** under hybrid threshold - smooth travelmoves
                                 crossing hybrid threshold - hard single knock on acceleration and deceleration

#### Additional Information

everything works on spreadcycle-only and stealthchop-only mode.
tried 12-24V, TMC2208/2209, from low to high acceleration, with or without s-curve,
other speeds for travelmoves and hybrid threshold

rado79, 2019-12-26:
new insights: 
enabled hybrid threshold - the maximum speed for hybrid thresdhold i can set on the controller menu settings is 255mm/s. 
now if i disable hybrid threshold and print only in stealthchop mode (XY), first i thought everything works fine. 
but, if travelspeed is more than 255mm/s it seems to jump in spreadcycle mode - same knocking, same noise, even though hybrid threshold is complete disabed. if enabled - knocking on threshold.
i tested with travelspeed 220, 240, 260, 280. working on driver max_current doesn`t matter (700-100mA), with or without s_curve, junction deviation, latest bugfix downloaded today.

ConstantijnCrijnen, 2019-12-27:
Can confirm the same exact issue with the bugfix from a couple of days back and earlier ones. 
Hardware: TMC2209, skr1.3, 24V, cartesian motion system.

sjasonsmith, 2019-12-27:
@ConstantijnCrijnen, can you add your configs? A set of Cartesian configs will be easier for most people to look at or use to try to reproduce.

ConstantijnCrijnen, 2019-12-27:
@sjasonsmith Unfortunately I can't currently because I'm away on vacation but I got it by taking the stock ender 3 config and changing the steppers to tmc2209 and hybrid threshold enabled with stock settings. 
The printer I'm working on is not an ender 3 though. It's a custom printer. I just used the example config because the mechanics and dimensions are similar

rado79, 2020-01-07:
because this is an dublicate of #16025 i will close

closed 2020-01-07 by rado79