MarlinFirmware/Marlin #16330: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements

MartinMajewski, 2019-12-25:
Hi there,

so I am now trying to configure Marlin 2.0.1 (stable) for my BigTreeTech SKR V1.3 to use two stepper drivers for the Z-axis (one driver for each stepper motor). I find the documentation is bad at this point so I'm making my way from one out-dated tutorial to the other.

Currently, I am struggling to set the correct steps per unit for E1 because I get the sanity check errors, which were mentioned here a few times. However, I have definitely enabled / defined / uncommented the line `#define DISTINCT_E_FACTORS`.

Still, I get the message

> Marlin/src/HAL/HAL_LPC1768/../../core/../inc/SanityCheck.h:2266:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)

So I cannot build the firmware. It this a bug, or have I missed something?

Best wishes,
Martin

Vertabreak, 2019-12-25:
did you set esteps for 2 extrudes and not enable distinct e factors?

MartinMajewski, 2019-12-25:
I have now disabled the sanity check and it worked. From first glance, I think there is an issue with

> did you set esteps for 2 extrudes and not enable distinct e factors? configs to check would help.

I said in the first post, that I definitely enabled distinct e factors (line 785 of Configuration.h).

Vertabreak, 2019-12-25:
need to post configs to see where the problem is.

MartinMajewski, 2019-12-25:
So I disabled the sanity checks just to get the build process going. But setting E1 to the same values as Z hasn't changed anything.

Configuration.h attached!

[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/4000888/Configuration.h.zip)

MartinMajewski, 2019-12-25:
Both steppers react to a "Move-Z" command, but with different speeds.

Vertabreak, 2019-12-25:
noting some very odd / in places they shouldn't be i think that is the problem.

MartinMajewski, 2019-12-25:
This is the stock Plattform IO C++ plugin dependency in Visual Studio Code. It f*cks up the formatting badly. I haven't figured how to disable this. Marlin 2.0 was freshly downloaded like an hour ago.

Vertabreak, 2019-12-25:
removed all of the erroneous characters and spacing.
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/4000907/Configuration.h.zip)

Vertabreak, 2019-12-25:
that's really weird vscode never did that to marlin on me.

MartinMajewski, 2019-12-25:
[image]

It's this option that "controls" the line wrapping of the C++ plugin. However, only "none" seems to work but it disables any kind of formatting support.

Vertabreak, 2019-12-25:
[image]

MartinMajewski, 2019-12-25:
With your file, I now get these kinds
[image]
 of error messages:

Vertabreak, 2019-12-25:
do you have platformio installed? all that's commented out.

MartinMajewski, 2019-12-25:
Okay, so the error message disappeared as soon as I switched from C++ to C in the bottom right corner.

However, even without all of those backslashes I have an issue building it:

[image]

Vertabreak, 2019-12-25:
i suspect it did that crazy formatting to the entire marlin source code.

Vertabreak, 2019-12-26:
[image]

MartinMajewski, 2019-12-26:
What do you mean with "also i don't see platformio anywhere on the screen"?
It is clearly there on the left-hand side. What you have opened is just the homescreen. I can make it reappear by klicking on the ant head at the left, then at the Quick Access menu -> PIO Home -> Open:

[image]

Vertabreak, 2019-12-26:
yea i didn't click on the photo its tiny on a 60inch tv for a monitor.

Vertabreak, 2019-12-26:
other then the messed up formatting it looked to be correct.

MartinMajewski, 2019-12-26:
I uninstalled VSC completely from my Mac... Killing everything and every extension.
Reinstalled VSC and PlattformIO. Made the bare minimum of configuration, namely added the "BOARD_BIGTREE_SKR_V1_3" motherboard definition, "Serial_Port_2 -1" and enabled "DISTINCT_E_FACTORS"

I then built the project once, without errors. Then I changed

`#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4000, 500, 4000 }`

1. The weird line breaks/wraps are gone!
2. Still the same issue with the "too many arguments" sanity check!

Vertabreak, 2019-12-26:
drop a copy of the new configs ill try it. i have a dual and tipple extruder setup that compiles that is very odd.

MartinMajewski, 2019-12-26:
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/4000995/Archive.zip)

Nothing else was changed!

Vertabreak, 2019-12-26:
[image]
compiled with 6 stepper drivers.
gonna try yours.

randellhodges, 2019-12-26:
I'm not sure you can use different step per unit for dual z steppers. If you look at the static_assert that is failing, it looks at `XYZE_N`, which, following the define chain, it basically adds the `#define EXTRUDERS` from you configuration.h.

Are the steppers the same? I am using dual z stepps, single x/y and single e.

https://github.com/randellhodges/Marlin/tree/skr-1.3/Marlin

Check out the configuration files there.

MartinMajewski, 2019-12-26:
@randellhodges That's what I was looking for. Thanks, I will look at your config tomorrow in the morning (it's 2 am here already).

However, having had the settings of the step left alone, both steppers (same model) driven by the same drivers (tmc2209) behaved differently.

How Are the stepper drivers even assigned to the axis? Where is it stated that E1 will be Z2 after enabling the Dual Z option?

Vertabreak, 2019-12-26:
[image]

MartinMajewski, 2019-12-26:
> i think this is what your looking for for dual z steppers.

I know this option and it is enabled (except now for the bare minimum config to try out the DISTINCT_E_FACTORS issue). Z and E1 react to move commands for the Z-axis simultaneously. However, both steppers move at different speeds.

randellhodges, 2019-12-26:
> How Are the stepper drivers even assigned to the axis? Where is it stated that E1 will be Z2 after enabling the Dual Z option?
I am not certain. I did a quick search and I didn't see anything. I think it might just be done `automatically`.

When you review my configs, once you set things up, 99% of things will now be Z2. There are one or two things that are still configured as E1. I don't remember them off the top of my head. I only have E0, so if you see any non-default settings for anything E1 related, that would be it.

randellhodges, 2019-12-26:
> > i think this is what your looking for for dual z steppers.
> 
> I know this option and it is enabled (except now for the bare minimum config to try out the DISTINCT_E_FACTORS issue). Z and E1 react to move commands for the Z-axis simultaneously. However, both steppers move at different speeds.

You might have updated configs, but looking at the last one...

commented out `#define E1_DRIVER_TYPE TMC2209` 
set `#define Z2_DRIVER_TYPE TMC2209`

`INVERT_E1_DIR` this one is still used for Z2. Assuming they are wired exactly the same, I use
`#define INVERT_E1_DIR INVERT_Z_DIR`

Note that you will configure current and stuff in configuration_adv.h as Z2

Then make sure you set any hybrid threshold Z2 the same as Z1, microsteps, etc

MartinMajewski, 2019-12-26:
> commented out #define E1_DRIVER_TYPE TMC2209
> set #define Z2_DRIVER_TYPE TMC2209

Perfect! That did the trick for me! No need for the stepping adjustment. Thanks!

I thought, that the Z2_DRIVER_TYPE is a setting for boards with dedicated two Z drivers. But actually it seems, that E1 is mapped automatically to Z2 as soon as the Dual-Z option is enabled.
This is so confusing and badly documented. Which steppers are mapped to what if you also enable Dual-X and Dual-Y??

Anyway, at least both steppers are running exactly the same now. Thanks a lot to you all for your support and patience.

closed 2019-12-26 by MartinMajewski