MarlinFirmware/Marlin #16331: [BUG] (Display not working correctly, on MKS Gen 1.4 Board)

aurion55, 2019-12-26:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4000971/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4000972/Configuration_adv.zip)

aurion55, 2019-12-26:
I've complie with No error so got me, I'm using
 #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

aurion55, 2019-12-26:
If I go back to a prevous version of my Marlin I get all these error's below.
Arduino: 1.8.10 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\src\module\endstops.cpp:36:0:

c:\users\gp73\appdata\local\temp\arduino_build_577121\sketch\src\hal\hal_avr\endstop_interrupts.h: In function 'void setup_endstop_interrupts()':

c:\users\gp73\appdata\local\temp\arduino_build_577121\sketch\src\hal\hal_avr\endstop_interrupts.h:142:7: error: non-constant condition for static assertion

       static_assert(digitalPinToPCICR(Y_MIN_PIN), "Y_MIN_PIN is not interrupt-capable");

       ^~~~~~~~~~~~~

In file included from sketch\src\module\endstops.cpp:36:0:

c:\users\gp73\appdata\local\temp\arduino_build_577121\sketch\src\hal\hal_avr\endstop_interrupts.h:57:57: error: reinterpret_cast from integer to pointer

                                     WITHIN(p, 62, 69) ? &PCICR : nullptr )

                                                         ^

c:\users\gp73\appdata\local\temp\arduino_build_577121\sketch\src\hal\hal_avr\endstop_interrupts.h:142:21: note: in expansion of macro 'digitalPinToPCICR'

       static_assert(digitalPinToPCICR(Y_MIN_PIN), "Y_MIN_PIN is not interrupt-capable");

                     ^~~~~~~~~~~~~~~~~

Multiple libraries were found for "U8glib.h"
 Used: C:\Users\gp73\OneDrive\Documents\Arduino\libraries\U8glib
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

So how can I get back to running version of my Marlin now?

ellensp, 2019-12-26:
The error is very clear. "Y_MIN_PIN is not interrupt-capable" so you need to disable endstop interrupts. In Configuration.h comment out #define ENDSTOP_INTERRUPTS_FEATURE

ellensp, 2019-12-26:
As for the display this is an issue with really slow LCD's You need to add in some delays. See #16278

aurion55, 2019-12-26:
Thanks I give it a go.

closed 2019-12-26 by aurion55