MarlinFirmware/Marlin #16333: [404] There isn't a GitHub Pages site here

rflulling, 2019-12-26:
A few users have pointed out on facebook that some one erased, or turned off http://marlinfw.org/
Visiting this link, results in Git generating a 404 error page. 

https://github.com/MarlinFirmware/Marlin still points users to http://marlinfw.org/
So if the link changed, it has not been posted. 

I understand that this is a nonstandard post. That this breaks rules for reporting. Just looking to draw attention as fast as possible from those who need to know. Perhaps even those who may place a phone call and wake up those who need to know.

alexdresko, 2019-12-26:
I'm most of the way through flashing my firmware for the first time, and all of the documents I need to complete the update route through marlinfw.org which isn't working. Kinda in a panic at the moment. Just wanted to confirm this is happening, mostly.

rflulling, 2019-12-26:
Still down.

thisiskeithb, 2019-12-26:
See https://github.com/MarlinFirmware/MarlinDocumentation/issues/265

ellensp, 2019-12-26:
Is working now

rflulling, 2019-12-26:
Yes it looks as though some one is trying to update the site. 
It is back online. This ticket is closed.

closed 2019-12-26 by rflulling