MarlinFirmware/Marlin #16334: [FR] add support for thermistor type 99 in Marlin\src\lcd\thermistornames.h

nforceroh, 2019-12-26:
### Description

add support for thermistor type 99 in Marlin\src\lcd\thermistornames.h
something like 
#elif THERMISTOR_ID == 99
  #define THERMISTOR_NAME "User defined"

You can't compile unless this is defined if you use 99 as a custom table

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2019-12-26:
this is a bug. Missing  THERMISTOR_NAME for THERMISTOR_ID == 99 
99 is already used. Needs a better name than User defined "100k bed thermistor with a 10K pull-up resistor" according to https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/thermistor/thermistor_99.h#L25

boelle, 2020-01-07:
@ellensp is this still an bug? or has it been fixed?

boelle, 2020-01-14:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-14 by boelle