MarlinFirmware/Marlin #16339: [FR] pause printing after a layer is finished

PetriJJ, 2019-12-26:
It would be very useful to have the ability to pause print after a layer has been finished. 
**Problem:**
The problem is that if you pause at any point and turn off the device (and go to sleep).. then when you turn on the device again in the morning and try to resume printing, the resume can fail.
The Manual fix for this failure is to modify the .gcode file, but the problem is that the printing was paused mid layer.
**Proposed solution:**
A menu option to pause the print after a layer has been finished. This way if the resume fails, it's possible to resume printing at the same point later.
(menu option could be called: pause on layer change)

MarkMan0, 2019-12-26:
Problem is that the printer can't really tell the difference between a Z-hop and a new layer. 
But I think, it could be something, like _pause on next Z change_ and the printer would print to the LCD/Serial the current line in the gcode

closed 2020-06-30 by github-actions[bot]

thinkyhead, 2020-07-04:
Use you slicer to insert `M125` or `M600` at certain layers for pause or filament change.