MarlinFirmware/Marlin #16340: [BUG]( LCD SD-card 'Insert' Not Detected For MKS_SGEN_L Board )

SharePARTS, 2019-12-26:
mks sgen-L + mks 12864 mini.

I've already read about it. 
https://github.com/MarlinFirmware/Marlin/issues/16223

I want to add one more thing here.
If a micro sd card is plugged into the MKS SGEN-L board, the same problem still occurs.
If you remove the card from the board, it works fine.

Can you solve this problem?

borland1, 2019-12-26:
Why would you need to plug in a SD-card in a slot that is not enabled with Marlin?

For MKS_SGEN_L board, since both the ONBOARD and LCD SD-CARDs share the same SD_DETECT processor pin, it is not practical to enable both slots at the same time. The board designer must have considered this as a limitation.  Regardless of which slot is enabled by firmware, the ONBOARD SD-slot is still active by bootloader during power-up, so that firmware updates can be accomplished.

SharePARTS, 2019-12-27:
Thank you for your kind explanation.
I understand.
By default, my printer has a sealed board.  So I have SDcard plugged into the board and uploading firmware via usb connection.  In this environment, I use the SDcard on the LCD, so both are plugged in.
If it's not something that you can modify inside Marlin, let's consider it mechanically possible.

SharePARTS, 2019-12-27:
Thanks for your kind explanation.
I didn't know one SD_DETECT processor pin managed two slots.
Let's approach it the other way around.

closed 2019-12-27 by SharePARTS

thisiskeithb, 2019-12-27:
It'd be great if manufacturers would standardize LCD controller designs/interfaces *with the same pinouts*.