MarlinFirmware/Marlin #16341: [BUG] JG Aurorsa A5S erratic touchscreen behavior

ishtob, 2019-12-26:
### Bug Description

Compiled and uploaded lastest bugfix-2.0.x branch using example config for the JG Aurora A5S. After flashing, touchscreen behaving erratically. When pressing + or -, get additional presses in the opposite direction.

### My Configurations
[Marlin-A5S.zip](https://github.com/MarlinFirmware/Marlin/files/4002861/Marlin-A5S.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Copy configuration.h and Configuration_adv.h from /config/examples/JGAurorsa/A5S and paste it to /Marlin
2. compile using vscode
3. Upload via SD

**Expected behavior:** [What you expect to happen]
+ and - should only move you up or down once.

**Actual behavior:** [What actually happens]

After firmware completes, start seeing extra button presses when using the touchscreen.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tpruvot, 2019-12-30:
Try to comment these settings to let the default, or set to 2 instead of 1

// ENCODER SETTINGS
//
// This option overrides the default number of encoder pulses needed to
// produce one step. Should be increased for high-resolution encoders.
//
//#define ENCODER_PULSES_PER_STEP 1

//
// Use this option to override the number of step signals required to
// move between next/prev menu items.
//
//#define ENCODER_STEPS_PER_MENU_ITEM 1

ishtob, 2019-12-30:
cool, I will try this after the current print that's running is done (got another 36 hours to go)

boelle, 2020-01-07:
@ishtob did you try it?

boelle, 2020-01-14:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-14 by boelle