MarlinFirmware/Marlin #16343: [FR] Add "Redo" option in Load Filament menu

yet-another-average-joe, 2019-12-26:
### Description

Some extruders MUST be loaded using the stepper. It's nearly impossible to do by hand (Bontech BMG). But sometime, it could go wrong for some reason (filament jam in the extruder for example). Then we have 2 options : purge again and again, or abort printing.

### Feature Workflow

A "Redo" option in the menu, between "Purge More" and "Resume"

[image]

If needed, I could shoot a video showing the feature in use.

#### Additional Information

Needs :

- 2 lines in pause.h
- 8 lines in pause.cpp (not recursive, uses a hideous goto because of the polling loop)
- 1 line in menu_filament.cpp
- 1 line in every langage_xx.h (= 24 lines)

Problem : not sure how I could publish the code...
Anyway, it's so simple that maybe it is not worth. And most of all, I maybe did it wrong. But it works and it is handy.
Could also add "Unload", for filament removing within the load function.

Vertabreak, 2019-12-26:
you could have filed a pull request to see if this gets approved.

Roxy-3D, 2019-12-27:
'Redo' ????   It might make sense to say 'Unload'.  

At the point this message is displayed, the carriage is raised and parked.    If the filament was unloaded,  it could be manually reloaded with more 'Purge More' commands.   That would give the user full control over the filament change on the nozzle.

I think I would also argue if this implemented, the appearance of an 'Unload' at this point should be controlled by yet another Filament Change option in Configuration_adv.h.   The reason being, most people won't want this feature.

yet-another-average-joe, 2019-12-27:
I don't konw how to do a pull request ! I created my Git a couple months ago, and have problems with Git Desktop (probably paranoid firewall related).

Even if I do a pull request, it will NOT be approved, because it needs localization ; moreover, I did not implement HOST_PROMPT_SUPPORT (never used it, and don't know what to do...). And maybe it is too quick and too dirty !!! (took me an hour, coding and testing ; I am not familiar with Marlin internals and its large amount of macros, and it's only my 2nd hack)

See this as a proof of concept.

I did this :

- I cloned Marlin to my Git
- I uploaded (overwrited) the files I edited (cannot commit with Git Desktop at this time), so no history, diff or whatever :(

Files :

- src/lcd/language/ : all files (to be localized, except for english and french)
- src/lcd/menu/ : menu_filament.cpp
- src/feature/ : pause.h and pause.cpp

(it should compile for any language, but the menu will display "Redo")

all changes are easy to locate, and they are tiny.
For language files, I added (in all files) :

`/******************************************************************/
// for "Redo" menu option
  PROGMEM Language_Str MSG_FILAMENT_CHANGE_OPTION_REDO     = _UxGT("Redo");
/******************************************************************/
`

menu_filament.cpp : see line 238
In pause.h : see line 67
In pause.cpp : see line 159 and line 307

Changes were also surrounded with large comments.

Hope it could help.

yet-another-average-joe, 2019-12-27:
Cross posting !

No, not unload ! It is used in order to redo the loading sequence in case the filament does not engage in the extruder. It happened to me... Pulling the filament out by hand was not too difficult, but using "Purge more" at 3 mm/s in a 500mm bowden was really tedious !

yet-another-average-joe, 2019-12-27:
I see it's been flagged "Feature Request". Could I close the issue and delete the fork ?

Roxy-3D, 2019-12-27:
> Even if I do a pull request, it will NOT be approved, because it needs localization ; moreover, I did not implement HOST_PROMPT_SUPPORT

You don't need to worry about 'localization'.   Just add the text you need to the English file.    And the build process will use the English text if no specific text is available for a various language.    Very soon, the translator people will provide a proper translation for the text in each language.

> No, not unload ! It is used in order to redo the loading sequence in case the filament does not engage properly in the extruder. It happened to me... Pulling the filament out by hand was not too difficult, but using "Purge more" at 3 mm/s in a 500mm bowden was really tedious !

It sounds like you have the filament load and unload lengths defined poorly in Configuration_adv.h.    You should double check that you have reasonable lengths declared in the 'Advanced Pause' setup area of the file.

yet-another-average-joe, 2019-12-27:
Thanks for the information about the locale files !

My lengths are finely tuned. If you consider Configuration_adv.h file on the fork, this file is untouched. It is not the one I use. The values on my printer are very different (and I do not forget to do M502 M500 afetr flashing). 30 mm for filament catching @ 6mm/s, 460mm @ 50mm/s for fast load, and 50mm @ 3mm/s for purging, after measuring filament length needed to reach the nozzle, and some tuning. It works great if nothing goes wrong when loading the extruder.

I created the fork only to show the few lines I added for the feature, all other files are original marlin sources.

closed 2019-12-27 by yet-another-average-joe