MarlinFirmware/Marlin #16347: [BUG] Temperatur alway BANG_BANG?

TAOGde, 2019-12-26:
### Bug Description

I update to the latest 2.0.1 version. after preheat my noodle I saw that the PID settings not working. it looks always like BANG_BANG. No Autotune for PIDD has helped so far.
The setting for PID enabled is on at default in marlin.

### My Configurations

[Archiv.zip](https://github.com/MarlinFirmware/Marlin/files/4003611/Archiv.zip)

### Steps to Reproduce

Start compiling with latest Arduino.
turned PID on and off with the same result

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

[image]
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2019-12-27:
Did you transfer your old PID settings when you upgraded? You may also still need to run a [PID Autotune](http://marlinfw.org/docs/gcode/M303.html).

ManuelMcLure, 2019-12-27:
There was a time when the way PID values were stored in EEPROM changed which resulted in bad PID behavior after updating Marlin. Check to make sure that Marlin really thinks the PIDs are what you specified. You might need to do a M502 to reset the EEPROM.

TAOGde, 2019-12-27:
> Did you transfer your old PID settings when you upgraded? You may also still need to run a [PID Autotune](http://marlinfw.org/docs/gcode/M303.html).

> There was a time when the way PID values were stored in EEPROM changed which resulted in bad PID behavior after updating Marlin. Check to make sure that Marlin really thinks the PIDs are what you specified. You might need to do a M502 to reset the EEPROM.

i do a fresh install and start PID Autotune with storing the settings in the firmware for hardcode settings and store it in the eeprom.

booth are the same result.

tatusah, 2019-12-27:
Are you absolutely sure you compiled the firmware with the PID temp enabled because the configurations you have linked show the PIDTEMP disabled?

TAOGde, 2019-12-27:
> Are you absolutely sure you compiled the firmware with the PID temp enabled because the configurations you have linked show the PIDTEMP disabled?

Oh this is my test then. I tried with PID on and off in the config. With on I make the pidtune and always the temperature is +-5 Celsius to the set Temperatur. 

With the old firmware all was fine. Pidtune and the Temperatur was Max. +-0,5  Celsius

tatusah, 2019-12-28:
Ok, so then it's not using BANG_BANG always if you have it set correctly. What was the old firmware version you used before and did you use the old configuration files? The configuration files you uploaded show PID_K1 value of 0.90 whereas current Marlin default example configs have value of 0.95. There was also other things missing from the configuration files that seem to indicate the config files might not be based on the actual 2.0.1 release. The comments for PID_K1 tell that it's a smoothing factor so I would try with the example value of 0.95 to see if this would fix the rather high oscillation of the temperature.

TAOGde, 2019-12-28:
So, I try a new config from the default drawer.
Setting my PINs and Stepper. Flashed the printer, reset the eeprom, start the PIDAUTOTUNE.
After that I saved the settings in eeprom, load the eeprom and start the tooltemperature to 230°C
as you can see in the curve the autotune works but the temperature after the tune does a BANG-BANG.

i uploaded the new configuration also.

[image]
[Archiv.zip](https://github.com/MarlinFirmware/Marlin/files/4006964/Archiv.zip)

tatusah, 2019-12-28:
Alright, very interesting. Comparing the two config versions show the first using bed thermistor 12 and the later one has thermistor 1. I know that the issue concerns the hotend temperature but if there is changes now with the bed thermistor then I think it's worth checking whether hotend thermistor type was also changed by accident. What printer model do you have or is it totally diy? Do you know which type / can you compare with the old firmware configs which thermistor type you had set for hotend?

At least on different control boards (SKR mini e3 v1.2 & Creality v.1.1.3) the PID temp seems to be working accurately with Marlin 2.0.1. With no other reports of the kind of issue you are describing this leads me to believe this is caused by either hardware fault, connection problem or configuration error. Googling around I noticed there has been similar reports of temperature problems with your board MKS gen L v1.0 but they were caused by power line problems and faulty capacitors on the board.

Do you have any graphs of the temperature from the previous firmware version and what was the previous version you were using? Old configs available? Have you been able to check with M503 that the tuned PID settings are actually saved especially if you turn of the printer before checking?

TAOGde, 2019-12-28:
> Do you have any graphs of the temperature from the previous firmware version and what was the previous version you were using? Old configs available? Have you been able to check with M503 that the tuned PID settings are actually saved especially if you turn of the printer before checking?

I yoused today my beta 2.0.x firmware. the right PID temperature is in the right
[image]
 graph.

tatusah, 2019-12-28:
Good to know. I suppose the old firmware was then also able to keep the temperature stable on higher temperatures like the 230 degrees you were testing on the newer version? Did you compile the older firmware yourself or was precompiled from the manufacturer? What I'm after is that do you have some source telling what type of thermistors and heating elements the printer uses? 

Luckily I was able to find this [repository](https://github.com/alexxxdev/Ghost4/blob/master/Configuration.h) with configuration for the Ghost 4. I'm not sure the author of the repository is actually affiliated with the manufacturer or not but it's a source anyway. Interesting details from the configuration is these:

```
#define BANG_MAX 150     // Limits current to nozzle while in bang-bang mode; 255=full current
#define PID_MAX BANG_MAX // Limits current to nozzle while PID is active (see PID_FUNCTIONAL_RANGE below); 255=full current
#define PID_K1 0.95      // Smoothing factor within any PID loop
```

Also the example has TEMP_SENSOR values for hotend and bed type number 11 and the bed heating was set to use BANG BANG which often is enough because the mass of the bed makes it so that the heat doesn't actually oscillate that much even with on/off control and it also wouldn't be so demanding for a 8-bit control board processor. Main thing in the example is the BANG_MAX 150, which could indicate that the hotend heating cartridge could be so powerful that it makes sense to limit the max current so that it wouldn't cause the temperature overshooting and sawtooth oscillation. Too aggressive and powerful heating element could . And as you can see the same BANG_MAX is also used for the PID_MAX. So if the old firmware wasn't compiled by you and you don't know the configuration values in that old firmware I would also try adjusting the BANG_MAX and perhaps also the TEMP_SENSOR types.

Just out of curiosity I read through some of the PID related code and didn't come up with a Marlin related cause for your issues. If the previous firmware wasn't compiled by you and the above settings doesn't help fix the issue and there isn't actual information of the needed settings and components used in your printer I think you would then need to get this information from the manufacturer and/or have them provide with new Marlin firmware. Alternatively someone else with MKS gen L or similar board would need to reproduce the issue. I still have high expectations with lowering the bang max value.

thisiskeithb, 2019-12-29:
> Googling around I noticed there has been similar reports of temperature problems with your board MKS gen L v1.0 but they were caused by power line problems and faulty capacitors on the board.

I struggled with maintaining temps with a couple Gen-L boards because of board issues. It drove me bonkers until I had one Gen-L that worked fine and realized that it was a design/manufacturing flaw after finding more reports of the same issue online.

I've been running `bugfix-2.0.x` on my [highly modded](https://github.com/thisiskeithb/Marlin/tree/myskr) SKR 1.3'd "Ender-3" with PID for the hotend/bang-bang for the bed (running an SSR/AC-powered bed) and it's been rock solid, so I don't think it's broken on `LPC176x` at least:

[image]

[image]

tatusah, 2019-12-29:
Good, I actually noticed some of your previous issue posts @thisiskeithb about MKS gen L temperature. I have also both 8 and 32 bit boards working fine with PID temp on bugfix and actual 2.0.1 release. So this doesn't seem like an actual bug in the core code.

TAOGde, 2019-12-29:
> Good to know. I suppose the old firmware was then also able to keep the temperature stable on higher temperatures like the 230 degrees you were testing on the newer version? Did you compile the [...]

i compiled all my firmware by my self. the working one and the bang_bang version.
the working on is only the hex file left. the rest is overwritten with the actual version :(

I know the repo you found. with this temperature sensor settings I got very wrong real temperature. I used an external sensor to match the value in the printer with my external sensor and at this point I stay at type 1 in this section.

the type of sensor in this printer I not known.

TAOGde, 2019-12-29:
I startet testing with  fresh installation and setting up every thing step by step with testing the temperature settings.
at the point "#define EEPROM_SETTINGS     // Persistent storage with M500 and M501" the bang_bang starts.
[image]

tatusah, 2019-12-29:
You shouldn't need to use M501 as it's for cancelling unsaved settings and reverting to the last saved settings. I would use M303 for autotuning to desired temperature and set those values to configuration.h and flash the firmware again. Then M502 & M500. After that M503 to confirm that the PID settings you placed in the configuration are actually loaded.

TAOGde, 2019-12-29:
> You shouldn't need to use M501 as it's for cancelling unsaved settings and reverting to the last saved settings. I would use M303 for autotuning to desired temperature and set those values to [...]

and that is the problem. I do this ^^ and the temperature goes bang_bang here. only without this setting in the source code all works like a charm before.

lrpirlet, 2019-12-29:
Well, when everything fails, it may be better going to the principles…

I suggest you have a look at this [https://en.wikipedia.org/wiki/PID_controller](https://en.wikipedia.org/wiki/PID_controller)... Have a close look in the **Manual tuning** paragraph at the animated curve and notice how the curve evolve with the 3 parameters…

Now, in the configuration.h file notice and follow this link [https://reprap.org/wiki/PID_Tuning](https://reprap.org/wiki/PID_Tuning)... Notice the **Modifying Marlin Autotune parameters** paragraph

It so happens that I had a lot of overshoot and a badly oscillating temp (I thought it was bang-bang)… Using the following did solve MY problem… maybe it helps.

> You can transform these "Classic" parameters into the Zeigler-Nichols "Some Overshoot" set with: 
>     Kp=Kp_classic*0.55
>     Ki=Ki_classic
>     Kd=Kd_classic*8/3

Enjoy.

AnHardt, 2019-12-29:
https://user-images.githubusercontent.com/43980652/71548154-6a21ca80-29aa-11ea-8d4c-6a625c9353f5.png
The above graph from autotune clearly shows - not enough iterations.
Target is to get about the same times above and below the target temperature.
If you can't get there with much more iterations, lower the PID_MAX.

TAOGde, 2019-12-29:
so, i test it again with M500 M501 and M502 commands after firmware flashing.
Hardcode load from firmware, writing to eprom and loading from it and the BANG_BANG is on. When i turn off the eeprom storage in the firmware ALL is fine and the PIDtemperature works well.

boelle, 2020-01-07:
is this a bug or a support question?

boelle, 2020-01-10:
> When i turn off the eeprom storage in the firmware ALL is fine and the PIDtemperature works well.

i assume this means it solved.... if not we can reopen if this is a bug and not support help

closed 2020-01-10 by boelle