MarlinFirmware/Marlin #16348: [BUG] G29 is probing in the wrong position

rmcbc, 2019-12-26:
### Bug Description

I'm using bugfix-2.0.x and the problem seems to be  in this commit: #15929 

G29 probes the points in the wrong position.

### My Configurations

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/4003650/Archive.zip)

### Steps to Reproduce

1. Send G28
2. Send G29

**Expected behavior:**

Probe the 9 points of the bed all around 30 positions from bed limits.

**Actual behavior:**

Is probing the 9 points much more than 30 positions from bed limits (too near the center of the bed)

rmcbc, 2019-12-27:
Fixed changing the file "G29.CPP"

[G29.cpp.zip](https://github.com/MarlinFirmware/Marlin/files/4006413/G29.cpp.zip)

Please search for "#if ENABLED(NOZZLE_AS_PROBE)"

InsanityAutomation, 2019-12-27:
Ill go over the action it takes there again, but the theory was the the inset would be honored for everything in the min / max position calculation. Prior it was possible to set points that did not honor configured limits. In reality those couple lines are a bugfix mixed with the feature extension to support lulzbot.

This would be expected behavior since you have 
#define MIN_PROBE_EDGE 30
in youre config. This number is a bit extreme and 5-10 is more common.

sjasonsmith, 2019-12-28:
@InsanityAutomation, I found a problem causing MIN_PROBE_EDGE to be doubled. I'm testing a fix right now.

sjasonsmith, 2019-12-28:
@rmcbc, would you be able to test my fix in PR #16367, to verify it solves your problem?

rmcbc, 2019-12-28:
@sjasonsmith for me your change works. Thanks.

dburr, 2019-12-29:
Experiencing the exact same problem, and the patch in #16367 solved it for me.

boelle, 2020-01-07:
seems this one is solved by #16367

closed 2020-01-07 by boelle