MarlinFirmware/Marlin #16350: Sensorless homing sensitivity values for TMC2130 drivers

SteenAndreassen, 2019-12-27:
Marlin 2.0.1, compiled on Visual Studio Code with PlatformIO Home: 3.1.0-beta.8 Core 4.1.1b6

According to the text in Marlin 2.0 (Configuration_adv.h) it says, that the sensitivity for TMC2130 values goes from -64 to 63, but it seems to be impossible to enter a negative number
If putting a negative number into the file Configuration_adv.h, compiling and later use command M122, a positive number is shown for "Stallguard thrs"

Neighter can a negative number be put in via Pronterface issuing the command: M914 X-2 Y-2

boelle, 2019-12-30:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

SteenAndreassen, 2020-01-02:
Sorry for the delay. Requested files attached
[Marlin_Config.zip](https://github.com/MarlinFirmware/Marlin/files/4015481/Marlin_Config.zip)

SteenAndreassen, 2020-01-02:
Opops wrong button

boelle, 2020-01-08:
is this the same if you use bugfix 2.0.x ? its updated more than 2.0.1

boelle, 2020-01-15:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-15 by boelle