MarlinFirmware/Marlin #16351: [BUG] (short description)

marcocm, 2019-12-27:
Hi all!
I have a skr 1.3 board with 4 tmc2209.
I am trying to use Z_DUAL_STEPPER_DRIVERS, where I connect the z2 stepper motor to E1 connection on the board.

I used in configuration.h 
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
#define Z2_DRIVER_TYPE TMC2209
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2209

 #define DEFAULT_AXIS_STEPS_PER_UNIT   { 100, 100, 400, 418.5450328 }

and in configuration_adv.h

#define Z_DUAL_STEPPER_DRIVERS

#define X_CURRENT       760 
#define X_CURRENT       760     
#define X_CURRENT       760     
#define X_CURRENT       760         

SENT: M122
READ: 		X	Y	Z	Z2	E
READ: Address		0	0	0	0	0
READ: Enabled		false	false	false	false	false
READ: Set current	760	760	760	500	900
READ: RMS current	734	734	734	489	887
READ: MAX current	1035	1035	1035	689	1251
READ: Run current	23/31	23/31	23/31	15/31	28/31
READ: Hold current	11/31	11/31	11/31	7/31	14/31
READ: CS actual	11/31	11/31	11/31	7/31	14/31
READ: PWM scale	43	45	48	9	16
READ: vsense		1=.18	1=.18	1=.18	1=.18	1=.18
READ: stealthChop	true	true	true	false	true
READ: msteps		16	16	16	16	16
READ: tstep		max	max	max	max	max
READ: pwm
READ: threshold	79	79	658	0	63
SENT: M105
READ: [mm/s]		100	100	3	-	30
READ: OT prewarn	false	false	false	false	false
READ: OT prewarn has
READ: been triggered	false	false	false	false	false
READ: off time	3	3	3	3	3
READ: blank time	24	24	24	24	24
READ: hysteresis
READ: -end		-1	-1	-1	-1	-1
READ: -start		1	1	1	1	1
READ: Stallguard thrs	130	130	0	0	0
READ: DRVSTATUS	X	Y	Z	Z2	E
READ: stst		*	*	*	*	*
READ: olb						
READ: ola						
READ: s2gb						
READ: s2ga						
READ: otpw						
READ: ot						
READ: 157C						
READ: 150C						
READ: 143C						
READ: 120C						
READ: s2vsa						
READ: s2vsb						
READ: Driver registers:
READ: 		X	0xC0:0B:00:00
READ: 		Y	0xC0:0B:00:00
READ: 		Z	0xC0:0B:00:00
READ: 		Z2	0x80:07:00:00
READ: 		E	0xC0:0E:00:00
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

randellhodges, 2019-12-27:
It might be reading what is in the EEPROM and not the configuration file changes.

Run `M503` and see what it says. For example, here is a snippet from mine:

```
Recv: echo:Stepper driver current:
Recv: echo:  M906 X1200 Y1200 Z1200
Recv: echo:  M906 I1 Z1200
Recv: echo:  M906 T0 E980
```

This is my 2nd z stepper: `M906 I1 Z1200`

You can either use that command to set it, or make note of any EEPROM settings you changed, then do a `M502` (factory reset) to reset EEPROM and `M500` to save that reset. You'll lose EEPROM only settings as everything will be replaced with defaults from the firmware.

marcocm, 2019-12-27:
I checked this values just after I updated the firmware. It is supposed to still read the EEPROM after I update the firmware?
I will try it when I get in touch with the printer again.

randellhodges, 2019-12-27:
Yes, unless the firmware changes the EEPROM structure it will continue to load EEPROM settings after a firmware update. That is why a lot of people suggest doing the EEPROM reset after every firmware update.

I don't unless I know I'm changing a value like you are, or I just run the gcode to set the value in the EEPROM (and also have the change in the firmware for later, when I do reset and forget).

marcocm, 2019-12-27:
It worked.. thank you!

marcocm, 2019-12-27:
It Worked, Thank You!

closed 2019-12-27 by marcocm