MarlinFirmware/Marlin #16353: [BUG] Stutter/Shake with Neopixel and SKR PRO 1.1

randellhodges, 2019-12-27:
### Bug Description

SKR PRO 1.1 with TMC2209 drivers in UART.
Neopixel on PE2 and they have their own 5v power supply.

I have them setup as case lights and the brightness was turned down. While it was printing, I used the LCD to turn up the brightness (I also have this in my version of the code https://github.com/MarlinFirmware/Marlin/pull/16345)

The part was printing the bottom layer and I noticed every time I turned the encoder knob to adjust the brightness the printer would studder/shake. I couldn't tell if a specific axis was affected or not.

I haven't yet tried moving the neopixel to a different pin to see if maybe there is some sort of interrupt conflict? Maybe PE2 isn't a good pin to use for this? I'm open to pin suggestions.

I also didn't check if just changing the colors had the same affect.

### My Configurations

https://github.com/randellhodges/Marlin/blob/507e9f39014c7c85076f04e14cdc178c345df44d/Marlin/Configuration.h
https://github.com/randellhodges/Marlin/blob/507e9f39014c7c85076f04e14cdc178c345df44d/Marlin/Configuration_adv.h

AnHardt, 2019-12-27:
215 NeoPixels in a row. And you really ask why it takes a while to update them?

They all get the same information. How about 5x43 or 6x36?

randellhodges, 2019-12-27:
Yes, they are case lights.

Are you saying that it takes a certain amount of time to update the lights and that blocks/prevents the printer from doing its real printer stuff? 

If that is the reason for the stutter then I'll do something different to drive them.

ellensp, 2019-12-28:
Yes that is what he is saying. If they are just display all the same colour he recommends connecting some in parallel. So instead of a 1 * 215 all in series you wire it up as 5 * 43 or 6 * 35 in parallel.   
The issue with that is the pin needs to be able to supply enough current for 5 or 6 chains...   Might need to put it through some sort of driver.

randellhodges, 2019-12-29:
Thanks @AnHardt for the info and thanks @ellensp for explaining.

I might move to an external controller, a pi, or a different solution all together. Never crossed my mind when I went this route that this could be an issue :)

Live and learn.

closed 2019-12-29 by randellhodges